csete/gpredict #12: Image formats not being recognized when being loaded

Analias, 2016-01-10:
Mac OS X 10.11.2 (El Capitan)
Using Homebrew ports of GDK2, GDK3, etc.

I wasn't sure where to document this or how to contact you, but I thought I would pass on this FYI for other folks who run into this problem.

After pulling gpredict from the repo and configuring to run with goocanvas 1.0, I was seeing the following errors when running gpredict:

```
pascal@Kusanagi <44> gpredict
2016/01/10 12:38:30   INFO  sat_log_init: Session started
2016/01/10 12:38:30  DEBUG  sat_cfg_load: Everything OK.
2016/01/10 12:38:31  ERROR  gtk-sat-map.c:1247: Error loading map file (Couldn't recognize the image file format for file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/maps/nasa-bmng-07_1024.jpg')
2016/01/10 12:38:31  ERROR  gtk-sat-map.c:1247: Error loading map file (Couldn't recognize the image file format for file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/maps/nasa-bmng-07_1024.jpg')

(gpredict:8736): Gtk-WARNING **: Error loading icon from file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/icons/gpredict-icon.png':
	Couldn't recognize the image file format for file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/icons/gpredict-icon.png'
(gpredict:7788): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(gpredict:7788): Gtk-WARNING **: Error loading icon from file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/icons/gpredict-sat-pref.png':
	Couldn't recognize the image file format for file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/icons/gpredict-sat-pref.png'
2016/01/10 12:38:35  ERROR  gtk-sat-map.c:1247: Error loading map file (Couldn't recognize the image file format for file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/maps/nasa-bmng-07_1024.jpg')

(gpredict:7788): Gtk-WARNING **: Error loading icon from file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/icons/gpredict-sat-pref.png':
	Couldn't recognize the image file format for file '/usr/local/Cellar/gpredict/1.3/share/pixmaps/gpredict/icons/gpredict-sat-pref.png'
```

The *file* command verified that the files were installed and were indeed the indicated format.  A bit of Googling returned the following solution:

```
# GDK_PIXBUF_MODULEDIR=/usr/local/lib/gdk-pixbuf-2.0/2.10.0/loaders gdk-pixbuf-query-loaders --update-cache
```

Problems with loading any images in any Gtk app have vanished.

csete, 2016-01-10:
Thanks for the info, but it seems to me that this is something that should be part of the package setup in homebrew, no?

Also note that I only use this repository as personal mirror and the official development place for gpredict is still on sourceforge and I don't know how many people will see it here.

Analias, 2016-01-10:
True, but I thought it should be documented somewhere other than my own notes and since it happened while trying to build gpredict and others might run into, I thought I would tell you.

BTW, gpredict and gqrx are great programs. I was a little broken hearted to see that I couldn’t just install gqrx with home-brew.  It has been moved to their boneyard due to the goocanvas changes.  I had to resort to building it by hand.  I don’t do a lot GUI development under unix/linux/osx so it’s all new territory for me. How hard do you think it would be to move to goocanvas 2?

In the past I have used Windows for my station control and logging. I would like to have a “native” solution for ham radio tools for the Mac. I don’t mind it being a port of Linux tools as long as I don’t have to run wine or a VM. The Mac is such a nice working environment.

-Freeman, N5FPP

closed 2016-01-10 by Analias

csete, 2016-01-10:
Porting to goocanvas 2 would mean porting from gtk 2 to gtk 3. It wouldn't be hard to port gpredict (not gqrx) to gtk 3 but it would require time that I do not have now. It has low priority since there is nothing wrong in having both versions installed in parallel.

csete, 2016-01-10:
By the way, I believe both gqrx and gpredict are fully supported in macports... not that I wish to start a discussion about homebrew vs. macports ;-)