csete/gpredict #32: Rig control after introduction of trsp download possibility

LongnoseRob, 2016-10-15:
With the introduction of the TRSP data the rig-control seems somehow broken. 
I already found out that for some sats trsp->downlow or downhigh is 0, but I havent discovered the details why this causes the rigcontrol not to show the proper frequenca of the selected transponder..

csete, 2016-10-15:
We are rewriting significant parts of the software and we are not done. Please consider gpredict HEAD experimental for now, thanks.

csete, 2016-10-15:
Can you please check if you still have problems with the latest code I pushed?

LongnoseRob, 2016-10-16:
@csete problem stil exsits, e.g. on FO-29 (Cat No 24278) where the V/U linear transponder is properly set, 
but the beacon entry are not working..

File 24278.trsp shows that for the beacons one entry is 0:
```
[robby@robquad trsp]$ cat 24278.trsp 
[Mode U CW Beacon]
UP_LOW=0
UP_HIGH=0
DOWN_LOW=435795000
DOWN_HIGH=0
MODE=0 baud:0.000000
INVERT=true

[Mode V/U Linear]
UP_LOW=145900000
UP_HIGH=146000000
DOWN_LOW=435800000
DOWN_HIGH=435900000
MODE=USB baud:0.000000
INVERT=true

[PE0SAT - CW Beacon]
UP_LOW=0
UP_HIGH=0
DOWN_LOW=435796520
DOWN_HIGH=0
MODE=CW baud:0.000000
INVERT=false

```

csete, 2016-10-16:
You have to re-download the trsp data.

LongnoseRob, 2016-10-16:
Right,

now its working fine.

Thanks!

closed 2016-10-16 by csete