csete/gpredict #35: The high/low frequency issue.

kukabu, 2016-10-18:
Hello

Some satellite in SatNOGS DB have only UP_HIGH/DOWN_HIGH and gpredict doesn't work with it.
Is it a bug in SatNOGS DB or gpredict?

kukabu, 2016-10-18:
hm... I found the same in local files https://github.com/csete/gpredict/blob/master/data/trsp/25509.trsp

LongnoseRob, 2016-10-18:
Did you try this whis the latest version?
I also found some issues with the satelite data and it was fixed already..

Maybe reimporting the TRSP data from SatNogs will help..

kukabu, 2016-10-18:
the latest version

csete, 2016-10-18:
Thanks for the info.

Previously trsp files were created manually and validated before becoming integral part of the package. Now we are in a new situation where data comes from external sources and we have to be able to handle all sorts of data.

I am in the process of cleaning up the side effects of the new trsp importer and also change the way we manage frequency data internally, but I am going to need some time to finish. But keep posting issues you find.

csete, 2016-10-18:
Please try it now. The code should now be robust for various missing parts of transponder data. You are also welcome to test it with manually edited files.

It was a late night commit but I hope I didn't ruin anything :-)

kukabu, 2016-10-19:
Fixed. Thank!

closed 2016-10-19 by kukabu