csete/gpredict #39: Fetching error from SatNOGS DB

ppapadeas, 2017-01-01:
On latest build I am trying to fetch SatNOGS DB transponder data and although the fetch of the JSON is successful, the parsing is not.

2017/01/02 00:20:37   INFO  Ready to fetch modes list from satnogs...
2017/01/02 00:20:37   INFO  modes_update_from_network: File to open /home/liknus/.config/Gpredict/trsp/modes.json
2017/01/02 00:20:37   INFO  modes_update_from_network: Successfully fetched https://db.satnogs.org/api/modes/?format=json
2017/01/02 00:20:37   INFO  modes_update_from_network: Fetched 1 files from network; updating...
2017/01/02 00:20:37   INFO  Ready to fetch transponder list from satnogs
2017/01/02 00:20:37   INFO  trsp_update_from_network: File to open /home/liknus/.config/Gpredict/trsp/transmitters.json
2017/01/02 00:20:38   INFO  trsp_update_from_network: Successfully fetched https://db.satnogs.org/api/transmitters/?format=json
2017/01/02 00:20:38   INFO  trsp_update_from_network: Fetched 1 files from network; updating...
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }
NXJSON PARSE ERROR (345): unexpected chars at }

csete, 2017-01-01:
Oh, I saw this once, but I simply don't remember what the issue was... It wasn't a gpredict issue.  Are you sure the file is correctly fetched?

csete, 2017-01-01:
It may have had to do with SSL, i.e. it only works with libcurl-openssl and not the other options.

ppapadeas, 2017-01-01:
@csete you mean which ssl I am using for build?

csete, 2017-01-01:
Sorry, I was confusing it with something else. Now I recalled where I have seen this issue:
https://community.satnogs.org/t/help-needed-gpredict-integration-with-satnogs-db/687/43
But I do not know why this error occurs.

ppapadeas, 2017-01-01:
Any ideas on how I can debug this?

ppapadeas, 2017-01-02:
Built the latest [commit](https://github.com/csete/gpredict/commit/0829add35fc748e3da3940cba69118242ccd0e7d) and it works like a charm. Closing this.

closed 2017-01-02 by ppapadeas