csete/gpredict #40: FT-817 support reported broken

csete, 2017-01-09:
https://sourceforge.net/p/gpredict/mailman/message/35593224/

csete, 2017-01-09:
Followup update through mail:

> I can pass audio without pressing the PTT however the radio does not come out of transmit with the second press of the space bar. It looks like the program comes out of engage 2 seconds after hitting the space bar the first time.
>  
> The only way to get it back into receive is to press and release the mic PTT switch.

csete, 2017-01-10:
The issue may have already been fixed in HEAD but we need to confirm.

---

> 2017/01/09 20:41:34|3|sat_log_init: Session started
2017/01/09 20:41:34|4|sat_cfg_load: Everything OK.
2017/01/09 20:41:34|1|qth_data_read: Error reading QTH type (Key file does not have key 'GPSDPORT').
2017/01/09 20:41:34|2|mod_cfg_get_integer_list_boolean: Failed to get integer list: Key file does not have group 'MAP'
2017/01/09 20:41:34|2|mod_cfg_get_integer_list_boolean: Failed to get integer list: Key file does not have group 'MAP'
2017/01/09 20:41:34|2|mod_cfg_get_integer_list_boolean: Failed to get integer list: Key file does not have group 'POLAR'
2017/01/09 20:41:34|2|mod_cfg_get_integer_list_boolean: Failed to get integer list: Key file does not have group 'POLAR'
2017/01/09 20:41:34|1|qth_data_read: Error reading QTH type (Key file does not have key 'GPSDPORT').
2017/01/09 20:41:35|2|mod_cfg_get_integer_list_boolean: Failed to get integer list: Key file does not have group 'MAP'
2017/01/09 20:41:35|2|mod_cfg_get_integer_list_boolean: Failed to get integer list: Key file does not have group 'MAP'
2017/01/09 20:41:35|2|mod_cfg_get_integer_list_boolean: Failed to get integer list: Key file does not have key 'HIDETRACKS'
2017/01/09 20:41:35|2|mod_cfg_get_integer_list_boolean: Failed to get integer list: Key file does not have key 'SHOWTRACKS'
2017/01/09 20:42:10|2|read_transponders: Error reading Mode U/V Beacon:UP_LOW from 7530.trsp. Using default.
2017/01/09 20:42:10|2|read_transponders: Error reading Mode U/V Beacon:UP_HIGH from 7530.trsp. Using default.
2017/01/09 20:42:10|2|read_transponders: Error reading Mode U/V Beacon:DOWN_HIGH from 7530.trsp. Using default.
2017/01/09 20:42:10|2|read_transponders: Error reading Mode U/V Beacon:INVERT from 7530.trsp. Assume non-inverting.
2017/01/09 20:42:10|2|read_transponders: Error reading Mode U Beacon:UP_LOW from 7530.trsp. Using default.
2017/01/09 20:42:10|2|read_transponders: Error reading Mode U Beacon:UP_HIGH from 7530.trsp. Using default.
2017/01/09 20:42:10|2|read_transponders: Error reading Mode U Beacon:DOWN_HIGH from 7530.trsp. Using default.
2017/01/09 20:42:10|2|read_transponders: Error reading Mode U Beacon:INVERT from 7530.trsp. Assume non-inverting.
2017/01/09 20:42:20|2|read_transponders: Error reading Mode V/U FM Voice:UP_HIGH from 27607.trsp. Using default.
2017/01/09 20:42:20|2|read_transponders: Error reading Mode V/U FM Voice:DOWN_HIGH from 27607.trsp. Using default.
2017/01/09 20:42:20|2|read_transponders: Error reading Mode V/U FM Voice:INVERT from 27607.trsp. Assume non-inverting.
2017/01/09 20:42:34|1|send_rigctld_command: SIZE ERROR 7 / 6
2017/01/09 20:42:34|1|../src/gtk-rig-ctrl.c: set_toggle rigctld returned error (RPRT -1
2017/01/09 20:42:34|1|)
2017/01/09 20:42:34|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:34|1|send_rigctld_command: SIZE ERROR 13 / 12
2017/01/09 20:42:35|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:35|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:35|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:36|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:36|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:36|1|send_rigctld_command: SIZE ERROR 13 / 12
2017/01/09 20:42:36|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:37|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:37|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:37|1|send_rigctld_command: SIZE ERROR 13 / 12
2017/01/09 20:42:37|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:38|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:38|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:39|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:39|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:39|1|send_rigctld_command: SIZE ERROR 13 / 12
2017/01/09 20:42:39|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:40|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:40|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:40|1|send_rigctld_command: SIZE ERROR 13 / 12
2017/01/09 20:42:40|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:41|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:41|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:42|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:42|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:42|1|send_rigctld_command: SIZE ERROR 13 / 12
2017/01/09 20:42:42|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:43|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:43|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:43|1|send_rigctld_command: SIZE ERROR 13 / 12
2017/01/09 20:42:43|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:44|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:44|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:45|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:45|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:45|1|send_rigctld_command: SIZE ERROR 13 / 12
2017/01/09 20:42:45|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:46|1|send_rigctld_command: SIZE ERROR 2 / 1
[... 12 lines cut ...]
2017/01/09 20:42:49|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:49|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:49|1|send_rigctld_command: SIZE ERROR 2 / 1
2017/01/09 20:42:49|1|send_rigctld_command: SIZE ERROR 13 / 12
[... cut ...]
2017/01/09 20:42:52|1|send_rigctld_command: rigctld port closed
2017/01/09 20:42:52|1|send_rigctld_command: SIZE ERROR -1 / 12
2017/01/09 20:42:52|1|send_rigctld_command: rigctld port closed
2017/01/09 20:42:52|1|send_rigctld_command: SIZE ERROR -1 / 6
2017/01/09 20:42:52|1|send_rigctld_command: rigctld port closed
2017/01/09 20:42:52|1|../src/gtk-rig-ctrl.c:close_rigctld_socket: Sent 2 bytes but sent -1.
2017/01/09 20:42:52|1|rig_ctrl_timeout_cb: MAX_ERROR_COUNT (5) reached. Disengaging device!

rutja76, 2020-04-10:
This issue is still present.

Pressing the space bar puts the FT-817 in transmit mode; however, after a couple of seconds the program comes out of engage. The radio remains in TX.

To go back to RX you can either briefly press and release the PTT or engage again Gpredict and press the space bar.

If you re-engage Gpredict and press the space bar the radio goes back to RX and Gpredict comes out of engage again after a couple of seconds.

So the problem is somehow related to pressing the space bar as Gpredict disengages in both cases (pressing space bar to TX and pressing it to go back to RX).

csete, 2020-04-10:
Thanks for the info @rutja76! I expect to have some time to work on this old and critical issue over the coming months so hopefully we can have it resolved soon.

pepassaco, 2022-08-18:
Hello! I can confirm that I'm currently suffering this problem too in the 2.3.37 version of GPredict. Has anyone found any workaround?