csete/gpredict #42: IC-910 support

csete, 2017-01-19:
We need to verify whether IC-910 support works as it should (i.e. full duplex) using the latest hamlib package. Fix it if it doesn't, and document how to set it up.

Access to an IC-910 is required to complete this taks.

jh4xsy, 2017-07-21:
At setup_split(), IC-910 enter into SPLIT mode. No need to use SPLIT/SAT mode.

At exec_duplex_tx_cycle() -> get/set_freq_toggle(),  We need to select Main VFO -> get/set Freq -> select  Sub VFO in needed.

I'm using IC-910D + hamlib 3.2 with following configuration / Radio type: Duplex TRX, VFO Up/Down: Main/Sub.

csete, 2017-07-22:
Thanks for the info @jh4xsy !

marwen01, 2018-05-22:
Hello,
I have an IC-910H and using hamlib 3.2
I cannot for my life get gpredict to control my radio in Duplex mode. 
Would love to replace SatPc32 with gpredict, but at the moment I am just pulling my hair.

Mode does not change and I am only able to kind of control one VFO at a time.
What am I doing wrong?

csete, 2018-05-22:
@marwen01 What do you mean what you are doing wrong? This is an open bug report with the topic is "IC-910 support is not working". Where is the confusion?

elitegreg, 2019-01-11:
I wonder if we should be talking to the hamlib guys about maybe making a set of commands for duplex radio operation since it seems some radios do it differently than others. I suspect lots of the problems (which appear to be caused by setting SPLIT) are due to radios like the TS-2000 which do satellites with split between VFO A/B.

marwen01, 2019-01-11:
Yes It could bea good Idea,

I will try to merge my code into and commit it to my fork when I find time, somehow I am always short of time.

It might not be the ultimate solution, but I have made lots of QSOs over many different birds with this version of gPredict controlling my IC-910, and at least might give some idea.

If you feel like Greg you have my email on qrz.com

Martin
A65DC

arodland, 2019-05-09:
I spoke to Martin, got a copy of his code, and rebased it against master. It's available as https://github.com/arodland/gpredict/tree/ic-910-hack . I don't think it's in a mergeable state as-is, but I'm using it successfully on my 910, and I would be willing to put some work in to bring it up to scratch. The chief difference is that instead of using an `F` command for main and `I` for sub, it uses `V Main` or `V Sub` to address each VFO and then `F` to change both.

There is another change included, which adds a button to the Radio Control dialog to change the mode between FM and SSB, which is nice (the panel buttons are inoperative while gpredict is driving the radio) but not really essential.

csete, 2019-05-11:
Thanks for this very useful info @arodland -- now I understand what the problem is and can think about what we can do to fix it without breaking support for other radios.

One solution I am thinking about is to add our own abstraction layer and use hamlib as a low-level backend.

elitegreg, 2019-06-14:
Since the issue is that Hamlib is broken with regard to handling the satellite mode on this rig and that code is shared in Hamlib, and thus difficult to change, I chose to replace Hamlib. I have written a drop-in replacement for rigctld in python that supports IC-910 (and probably all the other Icom satellite radios). Currently, the only commands implemented are those necessary to get it to work with GPredict. This is like "pre-alpha" state and needs a lot of work, but it will get me on the air with doppler correction for field day. Project is here: https://github.com/elitegreg/pyradioctl . In GPredict, the rig interface is added with "Duplex TRX" and "Sub Up / Main Down". Down the road, I'd like to add the other satellite radios from Yaesu and Kenwood (and flush out the rest of the Icoms). When that is done, I think adding a doppler correction command to this tcp server and patching Gpredict to use that command when using pyradioctl would be the best long term course of action, because as new rigs are added, the interface library can decide how to correctly apply doppler correction.

mikaelnousiainen, 2019-06-14:
@elitegreg I think it would still make more sense to fix this in Hamlib than to create a copy of Hamlib just for this purpose. After all, Hamlib quite widely in use. The next Hamlib release is going to be a major one (4.0), so it would be a perfect time to fix the broken IC-910 behavior right now. I know that many of the radio control backends for less common rigs have broken functionality -- I've just fix a large number of issues in Icom HF radio backends.

It seems that the IC-910 backend has some weird logic in `set_freq` command, which the rest of the Icom rigs don't have. I'm not sure why that code is there and if it should actually be removed, as the CI-V command set of IC-910 looks more or less the same (regarding VFO and frequency handling) as other recent Icom transceivers. It's worth having a look and contacting the maintainers if they'd happen to know why the piece of code is there.

P.S. Hamlib maintainers do welcome changes and they actively review and merge patches!

@csete This looks like an issue in Hamlib -- though it may help if gpredict is able to change/toggle levels/functions of rigs in certain cases.

elitegreg, 2019-06-14:
@mikaelnousiainen I appreciate the comments. At the of the day, I would spend more time fixing hamlib the rewriting what I need, so that's what I chose. If someone else wants to leverage my code, go for it, if not I'm happy with it. Hamlib seems to have suffered from organic growth and the code is an absolute mess. Combined with the fact that I hate writing C (I'm a C++ programmer by trade) and the risk of unintended consequences and the aforementioned time problem, I will personally not be fixing this issue. I do think Hamlib needs a method for doppler correction (single command that takes 2 int hz values). Gpredict shouldn't need to read the freq (in case the knob was turned), and update it and do that twice. The backend should provide the abstraction to do that, because, who knows, some new radio might come with that command in its command set!

dl7oap, 2019-11-23:
there was a bigger hamlib commit now for ic9700, ic9100 and ic910. 
so the main commands you need for satellite control (for example activating TONE, disable reaptershift DUP, ..) are now available when you use the newest master branch of hamlib for ic9700, ic9100 and ic910.

https://github.com/Hamlib/Hamlib/pull/143

[image]

The horizontal split via VFO A and VFO B (within Main) and the vertical split between Main (currentVFO) and Sub (currentVFO) is under progress. But it is hard and tricky. 

It seems that this topic is a problem for a lot of transceivers (for example TS-790), for a couple of years (found threads from year 2011 and older).
So for some transceivers hamlib seems to use MAIN as synonym for VFOA and SUB as synonym for VFO B. But for ICOM this is not true.

And in the moment i do not understand the methodic concept in general how hamlib command differ between this two (vertical and horizontal) split methodes. 

so for SPLIT by Main and Sub (S 1 Main ?) we do not want that the original SPLIT-Icom-original-functionn is activating.. when we ask for split by Main and Sub we want that SUB is active and SPLIT-Icom-original-function is off. 

maybe someone can explain me, how hamlib is design to solve this, but at the moment i think this is a problem of the design and it-architecture.

The easiest way seems to be to simply work with standard commands (V Sub, f, V Main), like in my python script, instead of using (S 1 Main, i), because hamlib command "i" is not really clear, what you will get (or how you will get > look at the flicker problem of icom transceivers).

N6RFM, 2020-03-16:
Has anyone made further progress on dealing with GPredict, Hamlib and the IC-910H?  Even when using the latest versions of these (great) packages, I'm unable to get doppler working properly in satmode or in split mode.  Any help, suggestions would be appreciated.  I can control the radio, but only on the uplink side.  Also, in split mode it makes no difference Adown/Bup or Bup/Adown.  Can adjust uplink frequency only. Any help, suggestions would be appreciated.  Thanks.

dl7oap, 2020-03-16:
Hi N6RFM, on the hamlib fork of Mike https://github.com/mdblack98/Hamlib/commits/master there is currently a lot of movement. So when you are able to compile daily the newest alpha version and can give test feedback to mike for IC910H that would be very welcome. Mike also have nightly builds for windows. Bye Andreas, DL7OAP

N6RFM, 2020-03-16:
Hi Andy,
Thanks for the kind suggestion!  I am pleased to report that sat mode now seems to be working for the IC-910H.  This is a major improvement, in addition to the great alternative of Mike's pyradioctl described above.  

However, I have observed that it only partially works when the rig is in split mode.  For example, VFO A and VFO B are being used.  They are fed with the desired frequencies (in the same band).  Let's say VFO A is for RX and VFO B is for TX.  When PTT is asserted, and at the end of the cycle when an update is being done, even with PTT still being asserted the radio will switch back to VFO A and continue to TX using VFO B.  Appears there needs to be a way to prevent VFO updates when PTT is asserted. I will pass this along to Mike.  Thanks again! 73, Bob

dl7oap, 2020-03-16:
Hi Bob, does the 910H have a VFO A and B in the main and also a VFO A and B the sub?
Or is VFO A identical to Main and VFO B to Sub?

For the IC9700 i using two different radio control setups in gpredict.
a) for U/V satellites i using "Duplex TRX" with sub as uplink, and main as downlink.
b) for simplex satellites (APRS frequency on ISS) i using "Duplex TRX" with VFO B as uplink and VFOA as downlink.

But i think the best way is to get in contact with Mike and creating a logfile 
rigctld -m 3081 -r /dev/ic9700a -s 19200 -Z -vvvvv  with the parameter -Z -vvvvv so he can see what is happen. Bye Andreas

N6RFM, 2020-03-16:
Hi Andreas,  

910H has a VFO A and B in the Main and also a VFO A and B the Sub.   My settings are the same as yours (will use -Z -vvvvv to gather data)
rigctld -m 3044 -r /dev/ttyUSB0 -c 0x60 -Z -vvvvv

If not already done so, you might also try setting up your IC9700 for APRS, and holding down PTT manually to see what happens.  I'm wondering if when the frequency is adjusted you will see VFO A or VFO B do something strange.  But I hope not!  

Cheers
Bob

dl7oap, 2020-03-16:
Hi, on ISS APRS  vfo a is active for RX and corrected by gpredict. there is no swapping to vfo b. only when pressing PTT, in just this moment the tx frequency is set and is corrected, too, during transmission. when you left PTT, it is switching back to vfo a (also with correction,again). so this works smooth and correct.

jharwinbarrozo, 2020-03-16:
> Hi Bob, does the 910H have a VFO A and B in the main and also a VFO A and B the sub?
> Or is VFO A identical to Main and VFO B to Sub?
> 
> For the IC9700 i using two different radio control setups in gpredict.
> a) for U/V satellites i using "Duplex TRX" with sub as uplink, and main as downlink. here no split is active.
> b) for simplex satellites (APRS frequency on ISS) i using "Duplex TRX" with VFO B as uplink and VFOA as downlink. here split is activ.
> 
> But i think the best way is to get in contact with Mike and creating a logfile
> rigctld -m 3081 -r /dev/ic9700a -s 19200 -Z -vvvvv with the parameter -Z -vvvvv so he can see what happens. Bye Andreas

I just finished compiling the latest snapshot hamlib 4.0 build of Mike from http://n0nb.users.sourceforge.net/ and was recompiled the current gpredict master. And I was able to replicate your a) and b) without any problem on my IC-910H.

My command for running rigctl is:
`rigctld -m 3044 -r /dev/ttyUSB0 -t 4532 -s 19200 &`

N6RFM, 2020-03-19:
Appreciate the suggestion Jharwin and sharing your own experience with the 910H.

Over the past few days, Mike Black W9MDB (Hamlib) has been very helpful (and patient) as we tried to sort this out.  Andreas DL7OAP has also kindly shared his experiences with the IC-9700, which provided a reference point. 

Can now report that doppler correction support for the IC-910H is working for me, including in spit mode!

Had been using the ppa verion, Gpredict 2.2.1-ubuntu 180.4 ~ppa.  Removed the ppa, and downloaded the latest GPredict version from source, and re-complied.  Now Version 2.3.58-c565.

Also using the latest version of Hamlib, again compiled from source.  https://github.com/Hamlib/Hamlib

As noted above, there continue to be frequent improvements to Hamlib, and clearly there are differences between the ppa and the latest version of GPredict.

So, thank you Mike (and the Hamlib team), Andreas and Jharwin for the help.  And of course, @csete and the GPredict team for GPredict.  These incredible resources.

73,

Bob
N6RFM

N6RFM, 2020-03-19:
I also want to acknowledge the work of Greg  https://github.com/elitegreg.  Greg was also helpful to me getting that running as an alternative to rigctld.  It only supports satmode at this time, but more features on possible in future.

His pyradioctl application is a drop in replacement for hamlib's rigctld. It only currently supports IC-910, and only enough commands for gpredict doppler correction.  It listen on port 4532. 

./pyradioctl -p /dev/your-serial-port-device -b 9600 -r IC-910

Thanks Greg.

N6RFM, 2020-04-01:
Just want to add an important point of clarification when using doppler correction with the IC-910H in split mode.  Do not enable split operation on the IC-910H physical panel itself.  From GPredict, only VFO A is used, and when TX is asserted, the proper TX frequency is written to VFO A.  upon going back to RX mode, the correct RX frequency is written back to VFO A.  

73 es GL all,

Bob
N6RFM

dl7oap, 2020-05-16:
IMHO this issue can be closed. with hamlib 4.0 it is solved. the only open issue here should be #217 which belongs to ic910, ic9100 an ic9700.

N6RFM, 2020-05-17:
Hi Andy,  That's fine.  For the current HamLib repo users should be aware that when changing from IC-910H SatMode to Split mode, the rigctrl daemon (rigctrld) may have to be restarted.  But for me, it's working.  Will continue to assist Mike to see if further improvements may be made, and track to  #217.  Again, kudos to all helping resolve this, and to Mike for working very hard to improve the HamLib functionality.  Hope to hear many of you on the birds.  73,  b.

csete, 2020-05-24:
Okay, closing this and we continue with issue #217.

closed 2020-05-24 by csete