csete/gpredict #43: Run TLE update in its own thread

csete, 2017-01-19:


csete, 2017-06-19:
Alternatively, we could use async calls in GIO.

yashi, 2020-12-08:
While looking at the code, it doesn't seem to have a way to set a specific proxy for `g_file_copy()`.  if you don't mind ditching the application specific proxy setting it's working right now on my env.

I've [asked about proxy on a forum](https://discourse.gnome.org/t/how-to-set-a-proxy-to-g-file-copy/4944) and been suggested to use libsoup.  Are you OK to add libsoup to dependencies?

csete, 2020-12-10:
Thanks for the heads-up. To be honest, I have been thinking about ditching the proxy setting anyway since I don't think anybody uses it (obviously, I can't know for sure, but I have never been able to test it).

I don't like the idea of adding a new dependency for this purpose. The advantage of using functions in Glib is that we could get rid of the libcurl dependency, but not if we would be replacing it with a new dependency :)

yashi, 2020-12-12:
Do you want to remove the proxy setting with v2.4? Would you mind creating an issue for it?

csete, 2020-12-19:
> Do you want to remove the proxy setting with v2.4? Would you mind creating an issue for it?

I have created an issue for it (#243) and I have sent out inquiries to see if anyone needs support for proxy servers.