csete/gpredict #47: possible TLE parsing problem

Sec42, 2017-02-05:
The current TLE for IRIDIUM 114 is this:

[Satellite]
VERSION=1.1
NAME=IRIDIUM 114
NICKNAME=IRIDIUM 114
TLE1=1 41923U 17003G   17034.00118558 -.13702181  00000-0 -27963+1 0  9995
TLE2=2 41923  86.6009  78.0456 0058911 258.0713  74.8913 14.67777747  2854

as of 2017-02-05 12:32:00 this produces an current altitude of 473km. 

My own calculations produce an Apogee and Perigree of
Ra [km]: 7087.81620154
Rp [km]: 7004.7952184
which put the satellite at somewhere between 625 and 710 km.

I suspect gpredict is making a mistake somewhere. - I verified with heavens-above.com which shows a completely different current position than gpredict.

csete, 2017-02-05:
Yes, some seems to be wrong with this. Unfortunately, I don't have time to look into this right now, sorry.

Sec42, 2017-02-05:
Thanks for your reply,

I debugged it further, and the issue is that the B* (drag) term of this TLE is just wrong (way to large). Most other tools ignore B*, but gpredict uses it, and thus arrives at implausibly low positions.

Possibly an option to disable B* in the calculations for powered satellites might be nice. One example use case would be if I wanted to predict positions in larger future, as it is to be expected that the satellite operator is going to keep the satellite on its intended orbit over longer times and will counteract the drag.

closed 2017-02-05 by Sec42

csete, 2017-02-05:
Thanks for the update. Is the satellite not in its final orbit yet?

Regarding the long term predictions, we have to remember that the SGP4/SDP4 algorithms used by gpredict are only approximations and require compatible input data that is updated regularly. I don't think it is suitable for long term predictions.

schneider42, 2017-02-06:
Iridium 114 is not in its final orbit yet. The satellites are being tested at the moment and will be raised to their final orbit in the coming weeks and months.

It looks to me like the B* term is tricky to measure and can have a significant amount of error if the measurement happens while the satellite maneuvers.

The TLE mentioned in the original posting contains B* = -2.7963 which is multiple magnitudes larger than usual. This caused SGP4 to decay the satellite very quickly.

The current TLE of Iridium 114 from http://www.heavens-above.com/orbit.aspx?satid=41923&lat=0&lng=0&loc=Unspecified&alt=0&tz=UCT is:
```
1 41923U 17003G   17037.22671398 -.01587209  00000-0 -38423+0 0  9999
2 41923  86.5344  76.7036 0037307 147.8937 197.0614 14.56172176  3325
```
This results in B* = -0.38423, which is still way to large in magnitude.

Compare this to Iridium 104 (http://www.heavens-above.com/orbit.aspx?satid=41922&lat=0&lng=0&loc=Unspecified&alt=0&tz=UCT):
```
1 41922U 17003F   17037.30253656  .00000018  00000-0 -91029-6 0  9994
2 41922  86.5273  76.6821 0002775 119.6580 240.4906 14.57104840  3329
```
Here B* = -0.91029E-6 which, I guess, is in the expected range.