csete/gpredict #61: autotrack does not update radio control

LongnoseRob, 2017-04-09:
with GPREDICT 1.3.460-6fc0-dirty:

activating autotrack before radio control still works as intended.
however starting radio control and then selecting autotrack does not update the sat selection in radio control.

could this be caused by the latest changes regarding threading?

csete, 2017-04-09:
The new threading is in the testing branch for now, so that's not it. I suspect this is just how it was implemented.

LongnoseRob, 2017-04-09:
@csete thanks for the explanation,

Please excuse the noise, I`ll close this issue.

csete, 2017-04-09:
Well, if it behaves the way you describe then it is a bug for sure. All I meant was that this is probably how it has been since the implementation (without knowing for sure), but we should most certainly look into fixing it.

DL4PD, 2017-12-15:
Tested in 2.0:
This is not reproducable for me.

csete, 2017-12-15:
Thanks @DL4PD - I guess you fixed it when you rewrote the controller ;-)

@LongnoseRob : any chance you could test it and confirm that it now works as it should?

LongnoseRob, 2017-12-16:
@csete undtunately I am not setup for testing this at the moment.
I would propose to close it

csete, 2017-12-16:
Ok, please let us know if you encounter it again.

closed 2017-12-16 by csete