csete/gpredict #62: IC-9100 support

csete, 2017-05-20:
Info received from Dimitris, VK1SV/VK2COW on May 12, 2017:

> I am helping the local university with setting up a
ground station for a QB50 satellite which will be in orbit soon. Our
satellite uses the same UHF frequency for both up and downlinking, of
course, the doppler needs to have the opposite sign when changing from
RX to TX. We use an Icom IC-9100 and I have set it up as Simplex TRX
and Read PTT. This works fine however there must be some race
condition because every now and then, Gpredict things that there has
been a dial change, and starts using that dial change resulting in a
random frequency jump. I found that this is more likely to happen when
using a digital mode (such as our satellite), where the time of
transmission is very short and transmissions happen very often.
> 
> I have downloaded the git master version and I have found in
gtk-rig-ctrl.c the if statement code that detects that change. I have
commented it out, for both rx and tx, and now it works perfectly. I
can spin the VFO dial and gpredict will immediately go back to the
correct frequency. If I want to make an offset change, I can do it
from the "Radio Control" window.
>
> Given there will be other IC-9100 users with UHF/UHF Up/Downlink
digital satellites, I think others will come across this problem.
Could I suggest you implement something like an "Ignore dial knob"
checkbox in the "Radio Control" window? The user then can check this
when using the Icom IC-9100 radio and not have that random frequency
change problem. Other users can leave that unchecked and it will work
as it used to.

csete, 2018-07-02:
There is a possible fix for this, unfortunately only in my inbox. I will try to turn it into a pull request for testing

lu9cbl, 2019-08-08:
Hi! the last version of gpredict has the fix fox this issue? Thanks

csete, 2019-08-14:
When a fix is available, the issue will be closed.

dl7oap, 2020-05-16:
IMHO this issue can be closed. with hamlib 4.0 it is solved. the only open issue here should be #217 which belongs to ic910, ic9100 an ic9700.

csete, 2020-05-17:
@dl7oap 

Okay, thanks. Can I ask you to summarize how gpredict and rigctld should be configured to have a working setup with the IC-9100? Something like https://github.com/csete/gpredict/blob/master/doc/notes/ft-847.txt ?

dl7oap, 2020-05-17:
Hi, the setup for gpredict and rigctld is the same as for the ic9700. Apart from the fact that you have to use a different hamlib modell number for the rigctld. 

would it be a solution to describe the 3 transceivers  (ic910, ic9100 and ic9700) in one file? 
most of the work is already done by vk2kyb with this pullrequest https://github.com/csete/gpredict/pull/220

csete, 2020-05-24:
> would it be a solution to describe the 3 transceivers (ic910, ic9100 and ic9700) in one file?

Yes, it sounds like a good idea, and perhaps rename the file to ic-9xxx.txt then?

db5fp, 2023-09-13:
Hi,

This question does not belong right to the IC-9100, but more or less how to get doppler correction work on a spefic radio.
I'm using an old TS-790 and I've tried different settings to obtain doppler correction for IO-117. 
So I was wondering how gpredict commands the radio to update VFOs and in witch order?

If this issue is missplaced tell me wehre to opne this discussion.

best regards

Frank