csete/gpredict #64: Right-click of satellite only works the first time

edbradshaw, 2017-06-13:
After launching, I am able to right click on any satellite and pull up the menu. Once I have done that, however, any other satellite I right-click still shows the menu for the first I selected.  ie. Start the application, right-click ISS, ISS footprint turns white and I can choose "Ground Track".  Now I right-click AO-85 but see the menu for ISS.

Ubuntu 16.04

LongnoseRob, 2017-06-13:
Could you please tell us which version of gpredict you are running?
(Help --> about gpredict)

I just tried to reproduce your described problem, but with my version of gpredict (GPREDICT 1.3.460-6fc0-dirty. built from git some weeks ago) it does not occur.
Means I get the right sattelite menu shown.

edbradshaw, 2017-06-13:
I am using 1.3.469-a5f0, which I had just cloned from Git last night.  I
just installed it on my work laptop this morning, which is the same Ubuntu
16.04.2, and it's working fine.  The only obvious difference is I'm using a
mouse at work and the trackpad at home.

I installed multiple versions last night, some from the PPA, some compiled,
and ran into the same problem. I do notice, on the system that is working,
that when I click on each satellite, the text and footprint turn white.  On
the one that isn't working, only the first satellite I click turns white.
What library is responsible for that function, I will remove and reinstall
 to see if that fixes it.

Thanks

Ed

csete, 2017-06-13:
This sounds very strange. I have never seen it except when one satellite covers the other, but as I understand we are not talking about that scenario.

Gpredict is drawing using libgoocanvas, which uses gtk2 and cairo, which uses X11, which uses graphics driver .... it could also be related to touchpad handling

Can you perhaps try using an external mouse instead?

edbradshaw, 2017-06-14:
I plugged in an external mouse and it worked just fine, though it takes an
extra click to get it working right again.  Keeping the mouse plugged in, I
continue to have issues with the trackpad.  I start using the mouse again
and it's working.

  It looks more like it's related to the touchpad and synaptics.  I will be
moving to one of the new Ubuntu releases in a few weeks and will try it
again then.

Thanks for getting back to me.

edbradshaw, 2017-08-03:
Sorry it's been so long to get back to you but I just discovered something
that may be of help.

I've installed Ubuntu Gnome 17.04 on the same laptop and see the same issue
happening, however, it looks like a setting ,I'm using, is causing the
issue.

For my trackpad settings, I normally choose "tap-to-click", which lets me
click on items by tapping anywhere on the trackpad.  I turned this feature
off and used just the manual click feature and it works just fine.  I've
also discovered that when "tap-to-click" is turned on, as long as I
manually click the objects, it works fine.

Ed

csete, 2017-08-04:
Hi Ed,

Thanks for the update. I am closing this issue since it is not a problem in gpredict, but feel free to keep discussing or post updates if you wish.

closed 2017-08-04 by csete