csete/gpredict #70: Translation issues

zxa, 2017-07-21:
I noticed that some strings that are translated on launchpad don't show up in gpredict. The problem seem to be that the strings are set as variables. Here are two files where this issue occurs:

/src/sat-pref-tle.c

Line:
#define ADBUT_TEXT  N_("Add new satellites to local database")
#define ADBUT_TIP   N_("Note that selecting this option may cause previously " \
                       "deleted satellites to be re-added to the database.")

/src/sat-pass-dialogs.c 
The lists at the beginning.

And there's a file missing in the potfiles.in:
gtk-sat-popup-common.c

Jan, DL2ZXA

csete, 2017-12-03:
Hi @zxa 

Normally, new translations from the Launchpad will not show up in gpredict until I actually merge them into the source tree. I do that occasionally, but haven't done it for a while. I will merge them again in a few days and then we can see whether there is a problem.

zxa, 2017-12-03:
Sorry, i wasn't clear enough: i'm speaking of my daily builds on launchpad. As i'm merging the launchpad translations into the daily builds, the translations are up to date in the ppa. But some strings don't appear in the builds, even when they are translated via the launchpad tool.

csete, 2017-12-03:
Hey, are you Jan Simon on the Launchpad who set up the daily builds?

I will take a look at why the translations don't show up.

zxa, 2017-12-03:
Yes i am ;-)

csete, 2017-12-03:
Ok, cool. I sent you a message through launchpad about the gtk+3 support which broke the builds, but you have probably noticed by now. New dependencies are in the readme, but let me know if you need more info.

closed 2026-08-16 by csete