csete/gpredict #71: Port Mismatch with gqrx remote control

vk4tec, 2017-08-22:
I was debugging a connection issue between gpredict and gqrx

The gqrx website and software says use port 7356

http://gqrx.dk/doc/remote-control

When I run up gpredict it is hard coded to port 4532

I used wireshark on localhost to spot this port mismatch

Can we align the two as to not confuse people

I am writing a gpredict to ICOM CIV program

Andrew VK4TEC

vk4tec, 2017-08-22:
my gpredict reports version 1.3 - is that the latest - I am trying to tell from GitHub 

Andrew

vk4tec, 2017-08-22:
MacBook:~ vk4tec$ sudo port uninstall gpredict
Password:
--->  Deactivating gpredict @1.3_2
--->  Cleaning gpredict
--->  Uninstalling gpredict @1.3_2
--->  Cleaning gpredict

Test for macports version - 23 Aug 2017

vk4tec, 2017-08-22:
MacBook:src vk4tec$ grep 4532 *
grep: nxjson: Is a directory
sat-pref-rig-editor.c:    gtk_spin_button_set_value(GTK_SPIN_BUTTON(port), 4532);
sat-pref-rig-editor.c:        gtk_spin_button_set_value(GTK_SPIN_BUTTON(port), 4532); /* hamlib default? */
sat-pref-rig-editor.c:    gtk_spin_button_set_value(GTK_SPIN_BUTTON(port), 4532);     /* hamlib default? */
sat-pref-rig.c:        .port = 4532,
sat-pref-rig.c:        .port = 4532,
sat-pref-rig.c:        .port = 4532,
grep: sgpsdp: Is a directory

csete, 2017-08-22:
You can use whatever port you want now

closed 2017-08-22 by csete