csete/gpredict #76: Add ICOM CIV control

vk4tec, 2017-09-25:
Because gpredict does not support serial port doppler correction for an ICOM radio I am forced to use macdoppler that does.

LongnoseRob, 2017-09-25:
Have you checked if rigctld could control your rig?
Then you could connect gpredict by network to rigctld..

(Sent from my mobile device)

csete, 2017-09-26:
Like @LongnoseRob said, gpredict uses hamlib to handle the radio specific interfacing, including ICOM CIV. So, if the hamlib rigctl utility can control your radio, then you can also control it from gpredict through  the rigctld daemon.

The user manual has more info about setting it up and testing it, see https://sourceforge.net/projects/gpredict/files/Gpredict/1.3/

closed 2017-10-17 by csete