csete/gpredict #80: Finish Gtk+ 3 migration

csete, 2017-12-02:
Replace all deprecated Gtk+ function calls.

gvanem, 2017-12-09:
In that regard, these should be applied:
```diff
--- a/gtk-sat-module-popup.c 2017-12-07 22:12:03
+++ b/gtk-sat-module-popup.c 2017-12-08 10:06:46
@@ -1261,7 +1271,7 @@
 #ifdef G_OS_WIN32
     /* Workaround for GTK+ bug # 169811 - "configure_event" is fired
        when the window is being maximized */
-    if (gdk_window_get_state(widget->window) & GDK_WINDOW_STATE_MAXIMIZED)
+    if (gdk_window_get_state(gtk_widget_get_window(widget)) & GDK_WINDOW_STATE_MAXIMIZED)
     {
         return FALSE;
     }
```
And:
```diff
--- a/main.c 2017-12-07 22:03:48
+++ b/main.c 2017-12-08 10:08:02
@@ -403,7 +403,7 @@
 #ifdef G_OS_WIN32
     /* Workaround for GTK+ bug # 169811 - "configure_event" is fired
        when the window is being maximized */
-    if (gdk_window_get_state(widget->window) & GDK_WINDOW_STATE_MAXIMIZED)
+    if (gdk_window_get_state(gtk_widget_get_window(widget)) & GDK_WINDOW_STATE_MAXIMIZED)
     {
         return FALSE;
     }
```

*PS*. Very good that development  have moved away from the awful SF. I've build GPredict using MSVC.-2017 So I have more patches. Would you accept those?

csete, 2017-12-09:
Thanks for the patch @gvanem - I have applied them. I'd be happy to accept patches that improve windows support, as long as they don't break linux support ;)

gvanem, 2017-12-09:
Okay, my main objection is this gnu-ism: `case '0' ... '9':` is 8 places. Can this be moved to `gpredict-utils.c` at-least (and maybe with a  `!defined __GNUC__` check)?

csete, 2017-12-09:
You will have to be more specific. Most of the code was written 10 years ago and I'm old enough to barely remember what I had for breakfast this morning ;-)

Also, please open a new issue for windows related fixes because this issue is for Gtk+ 3 migration and will be closed once Gtk+ 3 related compiler warnings have been fixed.

csete, 2017-12-09:
No need to move it, we can simply expand it.

csete, 2017-12-28:
DONE!

closed 2017-12-28 by csete

kukabu, 2017-12-29:
Hello

I see the warnings yet:
```
warning: ‘gtk_menu_popup’ is deprecated: Use '(gtk_menu_popup_at_widget, gtk_menu_popup_at_pointer, gtk_menu_popup_at_rect)' instead [-Wdeprecated-declarations]
warning: ‘gdk_screen_width’ is deprecated [-Wdeprecated-declarations]
warning: ‘gdk_screen_height’ is deprecated [-Wdeprecated-declarations]
```

csete, 2017-12-29:
Yes, I suppose they keep deprecating things in newer versions of Gtk+ but I can not see them on my development machine. I don't expect this to be a problem for a while.

kukabu, 2017-12-29:
Yes, these functions has been deprecated since 3.22.