csete/gpredict #82: Update user manual update for 2.x

csete, 2017-12-02:


mistermatt2u, 2017-12-06:
I would like to try to help with this. I'm a big fan of the application.

csete, 2017-12-06:
Hello @mistermatt2u 

That would be awesome as it would allow me to work on the code for the upcoming releases. I think the manual is mostly OK, but there is a general need to to update the contents and screenshots to be more recent. Of course, other improvements like layout are also welcome.

There is no pressing deadline, just would be nice to get updates to manual along the way as we release 2.1, 2.2, etc. FYI here is the current release plan:

* Version 2.0 will be released in a few days and will be the initial port to Gtk+ 3. It does contain some new features that are not documented
* Version 2.1 will finish the port to Gtk+ 3. At this point the UI should be fixed wrt. to Gtk+ 2 to 3 migration
* Version 2.2 will add the improved radio and rotator controllers

mistermatt2u, 2017-12-07:
Should I wait until 2.0 releases, before going thru the manual? Or do you think a build from master is close enough to the 2.0 final?

mistermatt2u, 2017-12-08:
So far I have:
* corrected some formatting
* validated and corrected all URLs and hyperlinks
* updated the copyright dates, version log 
* added bits about the SATNOGS db integration 

Next step is to build from master and see if there are any additional sections needed in the manual, or if any can be removed. 

It might make sense to commit what I have so far, and submit a PR. So at least the updates above are in before the 2.0 release. What do you think @csete ?

csete, 2017-12-08:
Yes, please make a PR so we can get your updates in. I expect to tag version 2.0 by the end of the week.

csete, 2017-12-11:
@mistermatt2u FYI, I have now created a dedicated label for issues related to user manual updates. This makes it easier to keep track of individual issues that come in. So far we have got #95 to consider.

mistermatt2u, 2017-12-11:
@csete Sounds great. I will put some time in this week.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

mistermatt2u, 2018-01-03:
Just noting here that PR #113 is a user manual update. Might be useful to track the various user manual related PRs here.

closed 2026-08-16 by csete