csete/gpredict #88: gmtime_r() copies to little

gvanem, 2017-12-09:
Compiling GPredict using *Clang-cl* gives this important warning:
```
sgpsdp/sgp_time.c(254,42) :  warning: 'memcpy' call operates on objects 
of type 'struct tm' while the size is based on a different type 'struct tm *' [-Wsizeof-pointer-memaccess]
   memcpy (result, local_result, sizeof (result));
           ~~~~~~                        ^~~~~~
sgpsdp/sgp_time.c(254,42) :  note: did you mean to dereference the argument to 'sizeof' (and multiply it by the number of elements)?
   memcpy (result, local_result, sizeof (result));
                                         ^~~~~~
```
Hence:
```diff
--- a/sgpsdp/sgp_time.c 2013-08-29 12:18:15
+++ b/sgpsdp/sgp_time.c 2017-12-09 09:56:56
@@ -251,7 +251,7 @@
    if (local_result == NULL || result == NULL)
      return NULL;

-   memcpy (result, local_result, sizeof (result));
+   memcpy (result, local_result, sizeof (*result));
    return result;
 } /*Procedure gmtime_r*/
```

gvanem, 2017-12-09:
Or a bit more compact:
```c
#ifdef WIN32
struct tm *
gmtime_r (const time_t *timer, struct tm *result)
{
   struct tm *local_result = gmtime (timer);

   if (local_result == NULL || result == NULL)
     return NULL;
   return memcpy (result, local_result, sizeof (*result));
}
#endif
```

csete, 2017-12-09:
Thanks for the fix. It is now applied.

closed 2017-12-09 by csete