csete/gpredict #90: Time Controller will not work in December

drotor, 2017-12-10:
When trying to change the date or time in the month of December using the Time Controller, the month jumps to January in the main window and the satellite positions change to the January positions set in the Time Controller which still shows the requested month of December.

csete, 2017-12-10:
Thanks for reporting, I will take a look. There is also another open issue with the time controller.

csete, 2017-12-16:
Hm, I can't reproduce this. Could you post a screenshot showing your main window and time controller?

kossy944, 2017-12-26:
I can confirm this issue too. When trying to select the next day of December in the time controller window, Gpredict month jumps to January as drotor commented previously. Please see screenshot.
[image]

csete, 2017-12-26:
THanks @kossy944, is this with the 2.0 release?

kossy944, 2017-12-26:
yes, version 2.0

[image]

csete, 2017-12-26:
Ok, thanks. I can now reproduce the error. The hint was 10 AM on your screenshot suggesting that you are using local time instead if UTC. And indeed, selecting "Show local time" in the configuration has this problem, but not UTC.

Thanks for the help, I will try and fix it for the next release.

kossy944, 2017-12-26:
Yes, UTC works as expected with the time-controller window when you select the next day. 
I will just switch back to UTC.
Thanks :-)

[image]

closed 2017-12-26 by csete

csete, 2017-12-26:
A workaround will be available in the next release.

commit c14bf120 (2017-12-26): Use correct month range when calculating UTC

Temporarily fixes #90 but also see issue #106.