csete/gpredict #91: Debian package shows UNKNOWN version in about box

csete, 2017-12-10:
[image]

mistermatt2u, 2017-12-10:
Separate but related, I'm seeing that the image on that window is not loading. Console error: `(gpredict:4299): Gtk-WARNING **: Error loading icon from file '/usr/local/share/pixmaps/gpredict/logos/gpredict_icon_color.svg':
        Couldn't recognize the image file format for file '/usr/local/share/pixmaps/gpredict/logos/gpredict_icon_color.svg'`

I don't see the "version UNKNOWN" problem. It is populating 2.0 as intended. This is on Ubuntu 14.04 (Windows BASH Shell).

I can make a separate bug report later. Does not seem urgent at this time. But just wanted to let you know that this "About" window might be having different behavior across platforms.

csete, 2017-12-10:
The problem with UNKNOWN version is when installing using the .deb packages. I think you may be running own compiled binary as suggested by the "/usr/local" path.

xtopheAmsat, 2017-12-11:
Dear, 

I have setup Gpredict 2.0 on Debian and have the same issue. 

    add-apt-repository ppa:gpredict-team/ppa
    apt-get update
    apt-get install gpredict

See : [ Amsat-f]( http://site.amsat-f.org/2017/12/10/nouvelle-version-de-gpredict-2-0-disponible/)

csete, 2017-12-12:
The problem is that we now use the git-version-gen script to generate a version number using the last git tag or a .tarball-version file generated and included in the "make dist" tarball.

However, using git archive to generate a source tarball results in a package that is neither a git repository nor does it have a .tarball-version file. Therefore git-version-gen will generate UNKNOWN version.

commit c5cea370 (2017-12-15): Propagate debian version

Fixes #91.
closed 2017-12-15 by csete