csete/gpredict #92: Help wanted with AppImage

csete, 2017-12-10:
I have a script that can generate an [AppImage](https://appimage.org/) from our [PPA](https://launchpad.net/~gpredict-team/+archive/ubuntu/ppa). It is based on the [official documentation](https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages).

The generated package runs but the program can not find any image or data files from the package.

I need somebody to dig into it and find out how to properly package.

probonopd, 2017-12-12:
Hi @csete - happy to help. Can you upload the AppImage for testing? Thanks.

probonopd, 2017-12-12:
In general, please refer to this section of the documentation:
https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages#no-hard-coded-paths

csete, 2017-12-12:
Hi @probonopd 

Thanks for getting in touch even before I contacted you :)

I think you are right regarding hard coded paths. It is a Gtk+ based application and uses autotools. The data and image paths are defined at compile time based on the --prefix=... option.

I will check the documentation you link to. I have uploaded the scripts I used and the AppImage here:
http://files.oz9aec.net/gpredict/temp/

probonopd, 2017-12-12:
Thanks @csete - does not run for me:

```
me@host:~$ Downloads/Gpredict-2.0.glibc-x86_64.AppImage 

** (gpredict:28105): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
me@host:~$ cat /etc/os-release 
PRETTY_NAME="Deepin 15"
NAME="Deepin"
VERSION_ID="15.5"
```

But yes, the first step is to get rid of the hardcoded absolute paths.

csete, 2018-01-21:
So, the [latest 2.2.1 release](https://github.com/csete/gpredict/releases/tag/v2.2.1) now has an AppImage where the paths have been fixed using the binary patch method. Since the hardcoded paths only exist three places I felt this was the easiest.

I have not yet looked into the runtime error you encountered.

tulipman, 2021-02-07:
Hi Csete.
I wonder if you are still active on the Gpredict software.

I don't see any activity since 2018.

Is it a 'dead horse' now??

I was interested in the program as it could (possibly control my Yaesu G-5500 El/Az rotor system, using just a Raspberry Pi.

Best regards

Robbert / PA3BKL (aka TA2IX)

AsciiWolf, 2024-08-02:
Any chance a new release of Gpredict could be made (ideally after merging [this PR](https://github.com/csete/gpredict/pull/363))?

The Gpredict 2.2.1 AppImage does not run on newer Linux distributions:
`symbol lookup error: /lib64/libgio-2.0.so.0: undefined symbol: g_module_open_full`