csete/gpredict #94: [Feature Request] Ground Track Improvements

tomswartz07, 2017-12-11:
Hey there:

Super glad to see that GPredict is getting some love lately.
I do have one feature request- I'd be glad to help out with implementation as well.

## Feature:
- Upon selection of a satellite in Map View, the ground track is shown, in addition to highlighting the footprint area
- If de-selecting a single satellite, the ground track is disabled
- While ground track is displayed, small indicators (arrow heads?) should indicate the direction of orbit

This feature makes it easy to quickly visualize the rough path of a satellite as it relates to the ground station, without having to resort to viewing the 'Sky At A Glance' window or inspecting individual polar pass info screens.

The direction of orbit indicators also allow for quick indication if it will be a N→S or S→N or any other combination of pass. 
This is especially useful when viewing the ground track 2+ orbits away for a polar orbiting satellite (such as NOAA 15/18/19), as it would otherwise be necessary to trace the path manually to determine.

These functions could be enabled or disabled as a user preference.

## Current process:
Currently, in order to enable the ground track on a specific satellite, you must right-click on the sat, select 'Ground Track'.
Then, repeat the process to disable the ground track.
To determine the direction of a polar-orbiting satellite's ground track, one must observe the current direction of orbit, then trace line forward.

tomswartz07, 2017-12-11:
Just doing some initial footwork here, with respect to the ground-track direction topic:

- It appears that the currently implemented line drawing function is GooCanvasPolyline.
- Docs indicate that the 'start' and 'end' can easily [add arrows](https://developer.gnome.org/goocanvas/stable/GooCanvasPolyline.html).
  - I think this is acceptable, since you can trace each single pass line across, rather than having to match it up with the previous orbit connection from the other side of the map. Ideally, the arrows could be placed at intervals along the line, but leveraging default behavior is simple enough. (No need to reinvent the wheel)
  - [It is indicated](https://developer.gnome.org/goocanvas/stable/GooCanvasPolyline.html#GooCanvasPolyline--end-arrow) that by default (if not specified) the start and end arrows are `false`
  - As such, it seems fairly trivial to implement as part of the [line drawing function](https://github.com/csete/gpredict/blob/master/src/gtk-sat-map-ground-track.c#L370).

Thoughts?

csete, 2017-12-11:
Hi @tomswartz07 

Thanks for the feedback. Unfortunately, I have to disappoint you... What has happened is that I have taken some time (that I did not have) and spent it on some required maintenance that was practically [forced down on me](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=785551). That work is still ongoing and the next 2-3 releases will be mostly clean-up and fixing/finishing existing functionality.

So any new feature that requires significant work will have to wait.

> Upon selection of a satellite in Map View, the ground track is shown, in addition to highlighting the footprint area
> If de-selecting a single satellite, the ground track is disabled

I see how this can be useful but I think it would be confusing together with the current "show ground track" option. I think the only way this might work is if we replace the existing "Show ground track" option with this way of showing the ground track. I will think about it.

> While ground track is displayed, small indicators (arrow heads?) should indicate the direction of orbit

At first I thought you meant a small arrow from the satellite pointing in the direction of moving, which is something I have actually thought about adding. But now I think you mean adding arrows along the ground track at regular intervals? I'm not sure how that would look...

Anyway, as I said, my plan right now is to spend the time that I don't have on clean-up and fixing existing stuff. You are of course more than welcome to play with the code and try things out, but please expect that any GUI code may have to be re-written after the port to Gtk+ 3 is complete.

tomswartz07, 2017-12-11:
Hey @csete:

Thanks for the feedback.
Understood on the next few releases. At the very least, I figured I would put out the suggestion/request to get it on the radar.

>I see how this can be useful but I think it would be confusing together with the current "show ground track" option. I think the only way this might work is if we replace the existing "Show ground track" option with this way of showing the ground track. I will think about it.

Thanks for the consideration. Totally understand.

>At first I thought you meant a small arrow from the satellite pointing in the direction of moving, which is something I have actually thought about adding. But now I think you mean adding arrows along the ground track at regular intervals? I'm not sure how that would look...

From the very first post in the feature request, I was imagining arrows all along the line.
However upon some cursory investigation into the gpredict codebase, I found that it seems to be relatively easy to add arrows at the beginning and end of the lines, instead, as you describe.
My previous comment was about using the line function parameters to define this behaviour.

For example, a quick mock-up:
[image]

If you're open to it, I'd be more than happy to play around with the code to see if this is easily implemented.

Thanks again for your work, I appreciate it.

csete, 2017-12-11:
You should play around with the code regardless of what I think; it's why it's there. On the other hand, I don't promise anything ;-)

srcejon, 2020-02-23:
I've implemented these suggestions here: https://github.com/csete/gpredict/pull/194