csete/gpredict #95: User's Manual - Request

csete, 2017-12-11:
Charles Phillips sent us the following request for user manual updates:

> On page 46 (of the new version) the discussion is about how to update from local files, could Alexandru add a specific note that you must select the directory and cannot select the enclosed file? I mainly run command line Unix applications and when I run a program that reads a file, I of course include the exact file name in my command. I am a Macintosh guy and when I do most things under the GUI with files I select the file I want to operate on.
> 
> I am still running 1.3 of course; I plan to move to 2.0 when it is available via MacPorts. 
> 
> But with Gpredict in this case - if you select the file inside the directory it does not work and there is not a useful error message. This is counterintuitive for me at least. Does it still work that way in version 2.0?
> 
> It would be more clear in the user's manual if it said that you cannot select the file. Another nice addition would be a message when you select the file, that you should select the directory. Another option is to allow the user to select either the directory or the file, I don't know how that might affect other versions.

Source: https://sourceforge.net/p/gpredict/mailman/message/36153156/

mistermatt2u, 2018-01-03:
Submitted PR #113 for updates to user manual section regarding updating orbital elements. In the case of build version 2.1.157-dab8, the browse window to select the folder does not allow selection of specific files. Hopefully that is true across platforms. In any case, the procedures were re-written to try to avoid any possible confusion.

csete, 2018-01-03:
It is only possible to select a directory in the current version. We will probably change this in a future update, though.