csete/gpredict #107: Win32 build of 2.193: Window scaling crashes gpredict

dl9sec, 2017-12-27:
When i try to make the gpredict window bigger, the program crashes.
This is the entry from the event logging of Win 10 1709b16299.125:

Protokollname: Application
Quelle:        Application Error
Datum:         27.12.2017 11:59:16
Ereignis-ID:   1000
Aufgabenkategorie:(100)
Ebene:         Fehler
Schlüsselwörter:Klassisch
Benutzer:      Nicht zutreffend
Computer:      NB-TG
Beschreibung:
Name der fehlerhaften Anwendung: gpredict.exe, Version: 2.1.0.93, Zeitstempel: 0x5a425201
Name des fehlerhaften Moduls: libpixman-1-0.dll, Version: 0.0.0.0, Zeitstempel: 0x529bca4b
Ausnahmecode: 0xc000041d
Fehleroffset: 0x0027095c
ID des fehlerhaften Prozesses: 0x3658
Startzeit der fehlerhaften Anwendung: 0x01d37eeadc4a059f
Pfad der fehlerhaften Anwendung: C:\ProgrammeUsr\gpredict\gpredict.exe
Pfad des fehlerhaften Moduls: C:\ProgrammeUsr\gpredict\libpixman-1-0.dll
Berichtskennung: 84e6b7da-6efc-452d-84bc-6e64b3cc7d86
Vollständiger Name des fehlerhaften Pakets: 
Anwendungs-ID, die relativ zum fehlerhaften Paket ist: 
Ereignis-XML:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2017-12-27T10:59:16.495536500Z" />
    <EventRecordID>4427</EventRecordID>
    <Channel>Application</Channel>
    <Computer>NB-TG</Computer>
    <Security />
  </System>
  <EventData>
    <Data>gpredict.exe</Data>
    <Data>2.1.0.93</Data>
    <Data>5a425201</Data>
    <Data>libpixman-1-0.dll</Data>
    <Data>0.0.0.0</Data>
    <Data>529bca4b</Data>
    <Data>c000041d</Data>
    <Data>0027095c</Data>
    <Data>3658</Data>
    <Data>01d37eeadc4a059f</Data>
    <Data>C:\ProgrammeUsr\gpredict\gpredict.exe</Data>
    <Data>C:\ProgrammeUsr\gpredict\libpixman-1-0.dll</Data>
    <Data>84e6b7da-6efc-452d-84bc-6e64b3cc7d86</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>

Maybe this helps...

csete, 2017-12-27:
@dl9sec I appreciate what you are trying to do, but gpredict for windows has not been released or thoroughly tested yet.

closed 2017-12-27 by csete

dl9sec, 2017-12-27:
Ok, no problem. Sorry for the disturbance...

csete, 2017-12-27:
Just keep an eye on test builds very soon in the new year ;)