csete/gpredict #110: Gpredict 2.x for Windows

csete, 2017-12-29:
I am starting this new thread to track status and progress of windows builds, since the original[ pull request](https://github.com/csete/gpredict/pull/84) is now merged and closed.

Status as of writing:

1. @phlash has updated the build files as well as some source files and can build a binary using mingw64 on Debian 9. He uses Gtk+/Glib binaries from Gnome project and these appear to be from 2012/2013. There is a [test binary package](https://github.com/csete/gpredict/pull/84#issuecomment-354363601) on Phil's webiste for testing, but this has stability issues, which can be related to the age of the libraries. We need newer libraries.
2. @gvanem has been building gpredict on windows using [his own Makefile](https://github.com/csete/gpredict/pull/84#issuecomment-353942888). This seems to support both MSYS2/MinGW and MSVC.
3. I have installed MSYS2/MinGW64 on windows following the [guidelines on the Gtk+ website](https://www.gtk.org/download/windows.php). The default instructions set up a 64 bit development environment and I have managed to build a [64 bit binary package](http://files.oz9aec.net/gpredict/temp/gpredict-win64-2.1.150.zip) that runs stable and looks like the default Gnome 3 theme (see my comment below).

csete, 2017-12-29:
I used the instruction on the Gtk+ website to setup MSYS2/MinGW. This has setup a more or less complete Unix environment and I could use the autotools method with a slightly modified configure.ac to build a binary.

The installation should also work using the updated Makefile as well as be able to build 32 bit binaries. I am going to try this next.

---

[image]

[image]

csete, 2017-12-30:
I continued with MSYS2/MinGW on windows and could install the 32 bit SDK along side the 64 bit SDK. I then copied the SDKs over to my Debian machine and could cross compile gpredict using the mingw-w64 cross compiler package.

So, using a standard MSYS2/MinGW installation we can build both 32 and 64 bit binaries on both windows and linux and using the latest Gtk+/Glib libraries.

I will now look into updating the runtime paths, icons, etc.

csete, 2017-12-30:
Latest build, now using the new application icon:

* [gpredict-win32-2.1.152.zip](http://files.oz9aec.net/gpredict/temp/gpredict-win32-2.1.152.zip)
* [gpredict-win64-2.1.152.zip](http://files.oz9aec.net/gpredict/temp/gpredict-win64-2.1.152.zip)

phlash, 2017-12-30:
Looks like MSYS2 is the way to go for up to date libraries, glad that sorted out the stability issue :)

Removing the experimental binary...

I spent a little while looking at how MSYS2 obtains it's libraries, with a view to avoiding the Windows installation + copy to Linux for a cross-build. It uses the Arch package manager pacman, with all the appropriate packages pulled from repo.msys2.org. This looks like it could work on an alien system such as Debian, sadly there is no port of pacman outside Arch or MSYS2 available for Debian-based distros. I will continue with building pacman from source to complete the experiment however.

phlash, 2017-12-30:
Experiment complete, it is possible to build pacman from source on Debian and use it to pull the appropriate MSYS2 i686/x86_64 dependencies for gpredict into local folders. I needed to re-compile the glib schemas (I suspect a native MSYS2 runs a post-fetch script for this):
```
cd /mingw32/share/glib-2.0/schemas/
/usr/lib/x86_64-linux-gnu/glib-2.0/glib-compile-schemas .
```
I also note that this approach appears to include a number of unnecessary libraries (eg: libpython2.7.so) that bloats the deployable ZIP by ~10M.

csete, 2018-01-02:
I think I have now fixed all remaining issues with the windows binary. As far as I can tell everything is working now.

[gpredict-win32-2.1.156.zip](http://files.oz9aec.net/gpredict/temp/gpredict-win32-2.1.156.zip)

I think this is sufficient for the 2.2 release and we can improve the readme and the makefile as we go.

demarcq, 2018-01-12:
Found few issue around web integration : 
- TLE Update from network return empty results (from file is ok)
- link to help don't work (no messages)
- about website link don't work (message : Could not show link)

csete, 2018-01-12:
@demarcq please file a separate bug. This is about building the binaries

srcejon, 2020-02-13:
I've just tried building git master for windows on msys2 using both mingw32 and mingw64 (x86_64) toolchains.

In order to get it to build & run on both, I needed to make a few changed to the Makefiles which are attached. 

Main changes are:
- including the png and jpeg loaders (jpeg seems to be needed for png to work - not sure why)
- including the icons/Adwaita/scalable directory in the distribution
- it no longer seems necessary to manually adjust pkg-config paths
- update compiler flags to support both 32-bit and 64-bit builds

[win32_build_patch.txt](https://github.com/csete/gpredict/files/4198123/win32_build_patch.txt)

W3AXL, 2021-06-04:
I've been trying to build the latest master on an MSYS2 install for my windows machine. I've configured the config.mk file to properly point to my /mingw64 folder, and it seems to be finding all the dependencies properly, but the build is still failing.

The last few lines I get when running the makefile are:

```
/opt/lib/gcc/i686-w64-mingw32/10.2.0/../../../../i686-w64-mingw32/bin/ld: sgp_in.o:sgp_in.c:(.text+0x26e): undefined reference to `g_ascii_strtod'
/opt/lib/gcc/i686-w64-mingw32/10.2.0/../../../../i686-w64-mingw32/bin/ld: sgp_in.o:sgp_in.c:(.text+0x2f5): undefined reference to `g_ascii_strtod'
/opt/lib/gcc/i686-w64-mingw32/10.2.0/../../../../i686-w64-mingw32/bin/ld: sgp_in.o:sgp_in.c:(.text+0x324): undefined reference to `g_ascii_strtod'
/opt/lib/gcc/i686-w64-mingw32/10.2.0/../../../../i686-w64-mingw32/bin/ld: sgp_in.o:sgp_in.c:(.text+0x393): undefined reference to `g_ascii_strtod'
/opt/lib/gcc/i686-w64-mingw32/10.2.0/../../../../i686-w64-mingw32/bin/ld: sgp_in.o:sgp_in.c:(.text+0x402): undefined reference to `g_ascii_strtod'
/opt/lib/gcc/i686-w64-mingw32/10.2.0/../../../../i686-w64-mingw32/bin/ld: sgp_in.o:sgp_in.c:(.text+0x458): more undefined references to `g_ascii_strtod' follow
collect2: error: ld returned 1 exit status
make: *** [Makefile:227: libsgpsdp.dll] Error 1
```

Since I'm no pro C developer I can't just start digging into the files to figure out what's wrong. I'm guessing it's something simple in the toolchain that's just not configured right. However I followed all the instructions on the GTK windows setup page, and additionally had to keep installing various versions of the mingw gcc packages until I found the one that contained the correct `i686-w64-mingw32-gcc` command.

Apace33, 2021-08-18:
> I've been trying to build the latest master on an MSYS2 install for my windows machine. I've configured the config.mk file to properly point to my /mingw64 folder, and it seems to be finding all the [...]

Same issue here, do you happen to remember which package it was?

curtiswilliams, 2024-06-27:
Anyone got a working windows build that’s somewhat up to date with the Linux version?

homes32, 2026-02-26:
@nmaas87 try https://github.com/csete/gpredict/pull/394 with the latest master. It removes the GooCanvas dependency.

nmaas87, 2026-11-14:
I opened up PR #448 and PR #449 - these two PRs do make building the Windows client possible again. First one is to fix all the small issues, second one is only adding the CI stuff in case one wants to build the windows client alongside in Github.

Full disclosure: These changes were created using qwen3.8-flash-next-iq3_s hosted locally on my own rig. I hate corpo AI and did not wanted to feed this wonderful program into thats throat. Also, I don't want to fund any of those corpos. So I am using my own RAM and rig ;). But if thats not acceptable, I will also understand and delete these PRs, no worries.

For those that finally want a build windows client of the latest commit, here you go: https://nico-maas.de/download/gpredict_2.7-windows_amd64.zip

csete, 2026-11-14:
Thanks for the two PRs @nmaas87. I have merged #449, which also included the commits from #448, so I think all is well. I also took the liberty of adding your 2.7 build to the release.

I am perfectly okay with using AI tools as long as they are supervised by human :)

closed 2026-11-14 by csete