csete/gpredict #111: Better alignment of list view columns

csete, 2017-12-29:
We should try to avoid awkward looking list views like the one on the screenshot below

[image]

jasonuher, 2018-07-06:
I have a branch that solves this by adding an empty column to the end of the list view.  This is not the most elegant solution, but in practice it ensures that the last visible "data" column always has a width set by QT's "auto width" decider.

The only real draw back from a user perspective is the niche case where "there are enough data columns to consume (the full width of the screen)+(the minimum width of the blank column)".  In this case there is a horizontal scroll bar added that is not technically needed.  In the other cases "data columns width less than widget width" and "data columns width large enough to create a scroll bar anyway" the behavior is un-noticable.

csete, 2018-07-06:
That might be part of the solution but it looks like there is more to be fixed here. For example, the alignment of the column header seems to be different from the alignment of the cell contents.

jasonuher, 2018-07-06:
> the alignment of the column header seems to be different from the alignment of the cell contents

If I understand the current code, this is on purpose (or, at least, for a known reason):

For events, there are two (non-identical) alignment arrays that are defined [here](https://github.com/csete/gpredict/blob/24e7f0b69c947ebea005eedd69fc4fe8031c206b/src/gtk-event-list.c#L70) that are used for the titles and data respectively.  If you wanted them to match up these would have to  be the same.

For the sat list you specify that the column's data should be set according to `SAT_LIST_COL_XALIGN` [here](https://github.com/csete/gpredict/blob/f95cc6ef8c2f93cf4aae8e37449cbfe2ac78ad2c/src/gtk-sat-list.c#L334), but then override the alignment of the column _header_  [here](https://github.com/csete/gpredict/blob/f95cc6ef8c2f93cf4aae8e37449cbfe2ac78ad2c/src/gtk-sat-list.c#L354).  If you wanted the headers to match the same alignment as the data field you would need to change line 354 to `gtk_tree_view_column_set_alignment(column, SAT_LIST_COL_XALIGN[i]);`

csete, 2018-07-06:
I don't remember what my intention was 15 years ago. There is a lot of legacy crap code in gpredict that needs to be cleaned up and the recent migration to Gtk+ 3 did not help much. Bottom line is that it looks stupid when the header is left aligned and the cell contents are right aligned.

jasonuher, 2018-07-09:
Do you want me to prepare a pull request that does these?

1. Consistently aligns the columns according to SAT_LIST_COLXALIGN
2. Adds an empty column to the end of the list to support the auto-width calculation