csete/gpredict #112: Homebrew formula missing

Vinz87, 2017-12-31:
I installed Homebrew to install gpredict 2.0 on OS X 10.11, as suggested in the home page, but the formula "brew install gpredict" doesn't work.
Was it removed for the new 2.0 update? Has it changed?

Vinz87, 2017-12-31:
Homebrew says what follows:
_gpredict was deleted from homebrew/core in commit ee7405d:
  Migrate some formula with hard-X11 dependencies.
  We have the homebrew-x11 tap for this reason so let's migrate everything
  without any reverse dependencies. Ideally I'd also like to migrate
  everything that's :recommended and the GTK formulae through there too
  (or to a dedicated homebrew-gtk)._

I tried googling it but I didn't find an answer: what's homebrew-x11? a different version of homebrew?

csete, 2017-12-31:
Sorry, but I have no idea. This is a bug tracker for gpredict, not homebrew. I have never had anything Mac OS X packaging related in gpredict repository, so you have to ask in the homebrew project.

closed 2017-12-31 by csete

csete, 2017-12-31:
I checked the project and can see what you mean now. They have [deleted gpredict](https://github.com/Homebrew/homebrew-x11/commit/10f6880be9cfd012669066f7e99e7bd076b71ebd) because they didn't want to keep the old version of goocanvas. I don't recall getting any notice from them when they did that, but its OK. I will just remove the reference from the home page.

You can still get [gpredict 2.0 from Macports](https://www.macports.org/ports.php?by=name&substr=gpredict). It may be old fashioned, but at least it is reliable.