csete/gpredict #115: Macports installation issue

Vinz87, 2018-01-12:
I installed gpredict through Macports via the command 
`sudo port install gpredict`

After it finishes, without errors, I launch the command `gpredict` and I get this error:
`(gpredict:18094): Gtk-WARNING **: cannot open display: `

Do I miss something to install?

csete, 2018-01-12:
I have no idea, sorry. Please note that we use this as a bug tracker not a tech support forum. We already have a mailing list and forum for tech support.

closed 2018-01-12 by csete

Vinz87, 2018-01-13:
well, I get the same error when I tried to compile it from source, on OS X El Capitan.

So it's not related to Macports, shouldn't it be reported as a possible bug?

csete, 2018-01-13:
You are still using the Gtk+ libraries installed and configured through macports aren't you?

You are welcome to report it as a bug here but this is not something I can fix in gpredict. For sure something went wrong with the installation or configuration of the toolkit. There are people already using gpredict via macports. You can find them on the forum and on the mailing list, see http://gpredict.oz9aec.net/User_Support.php

csete, 2018-01-13:
I think you need to have an X server installed, which may not be installed by default. Still, it's best to as those who have it running on the mailing list or forum.

Vinz87, 2018-01-13:
Update: I solved it installing XQuartz on my Mac.
Thank you anyway