csete/gpredict #116: web issue on windows ?

demarcq, 2018-01-12:
Found few issue around web integration on Windows 10 / 2.2 version :

=> TLE Update from network return empty results (from file is ok)
=> link to help don't work (no messages)
=> about website link don't work (message : Could not show link)

matt-nj4y, 2018-01-12:
Windows 10, v 2.2 binary downloaded from github.

the TLE update works if no gpredict.cfg file has been created with parameters for TLE's. Once I edit the TLE source list, and the .cfg file is created, updating from the network returns empty results. 

If I remove the [TLE] parameters from the .cfg file, I can once again download them via the network.

Error log:

[image]

csete, 2018-01-12:
@matt-nj4y  Are you editing the gpredict.cfg file manually or through the GUI?

matt-nj4y, 2018-01-12:
The first time, through the GUI. That is when it stopped working properly. Then, I edited the file in notepad++, removing the [TLM] block, and it started working again.

demarcq, 2018-01-12:
Confirming that I have the same issue and solving fetching removing the TLE block.

csete, 2018-01-12:
Did you try using the "Reset" button the the config page? It should have the same effect, unless there are even more bugs ;)

gvanem, 2018-01-20:
Confirmed here too. It's due to the `INTERNET_FLAG_RELOAD` flag; in `InternetOpenUrl()`. Using no flags fixes this `ERROR_INVALID_PARAMETER`.  (0x57 == 87 decimal).

phlash, 2018-01-20:
The error number is coming from my code here:

```
int win32_fetch(char *url, FILE *file, char *proxy, char *ua)
{
    HINTERNET hInt, hUrl;
    char buf[BUFLEN];
    DWORD nRead;

    if (proxy != NULL)
        hInt = InternetOpen(ua, INTERNET_OPEN_TYPE_PROXY, proxy, "<local>", 0);
    else
        hInt = InternetOpen(ua, INTERNET_OPEN_TYPE_DIRECT, NULL, NULL, 0);
    if (NULL == hInt)
        return 0x80000000 | (int)GetLastError();
```
The error itself (0x57) is 'Invalid parameter', so I guess the 'proxy' parameter is set to something odd once the config file has been saved? Note that the 'ua' parameter is a constant in calling code.

gvanem, 2018-01-20:
Confirmed. When disabling the proxy in `win32_fetch()`, all is okay. 
0x57 above is `87 == ERROR_INVALID_PARAMETER`.

Can be sure an  empty `PROXY=` in `gpredict.cfg`, is not returning `""`?

csete, 2018-01-20:
I think initially we return NULL, but after the user edits the settings we save `""` if the proxy URL is empty:
https://github.com/csete/gpredict/blob/master/src/sat-pref-tle.c#L445

I will update sat-pref-tle to delete the record instead, then I think no changes to the windows code is necessary?

csete, 2018-01-20:
@gvanem, could you check to see if the above commit is sufficient on windows?

gvanem, 2018-01-20:
That commit didn't help. Still the `ERROR_INVALID_PARAMETER`  from `InternetOpen()`.

csete, 2018-01-20:
So, it doesn't do anything unless you open preferences, make some change and apply with OK.

csete, 2018-01-20:
I guess it will be a good idea to check for empty string anyway:

```
diff --git a/win32/win32-fetch.c b/win32/win32-fetch.c
index a03bd4b..7120c33 100644
--- a/win32/win32-fetch.c
+++ b/win32/win32-fetch.c
@@ -31,6 +31,7 @@
 #include <windows.h>
 #include <wininet.h>
 #include <stdio.h>
+#include <string.h>
 #include "win32-fetch.h"
 
 #define BUFLEN 8192
@@ -41,7 +42,7 @@ int win32_fetch(char *url, FILE *file, char *proxy, char *ua)
     char buf[BUFLEN];
     DWORD nRead;
 
-    if (proxy != NULL)
+    if (proxy != NULL && strlen(proxy) > 0)
         hInt = InternetOpen(ua, INTERNET_OPEN_TYPE_PROXY, proxy, "<local>", 0);
     else
         hInt = InternetOpen(ua, INTERNET_OPEN_TYPE_DIRECT, NULL, NULL, 0);

```

csete, 2018-01-20:
Never mind, I decided to put the check in the caller and so the empty PROXY record is deleted as soon as TLE update is run.

csete, 2018-01-21:
The issue should be fixed in the 2.2.1 release. Feel free to reopen if there is still an issue.

closed 2018-01-21 by csete