csete/gpredict #117: .tarball-version missing in release archives

sielicki, 2018-01-19:
Version appears as 'UNKNOWN' as a result.

sielicki, 2018-01-19:
See https://github.com/csete/gpredict/blob/master/git-version-gen#L69-L135

`tar tvf v2.2.tar.gz  | grep .tarball` returns nothing. As far back as v1.3 I cannot find this file on the github release archives. As a result, 'About gpredict` will report version of UNKNOWN.

I suspect that when you prepared release archives on sourceforge, you were doing so manually. Now that you're using git tags to have github generate the release archives, this file needs to be present in the repository when the release tag is made.

I ran into this issue when preparing a flatpak-builder file for gpredict and building from the github archive. Using the git tag instead works fine and the version reports correctly.

csete, 2018-01-19:
The release tarball is https://github.com/csete/gpredict/releases/download/v2.2/gpredict-2.2.tar.bz2 and contains a .tarball-version. The other packages are automatically generated by github and I have no idea how to disable them.

If you want to use the code from git please do a git clone and don't use that junk generated by github.

sielicki, 2018-01-19:
Don't know how I missed that, sorry for the confusion. That's an unfortunate feature of github.

One possible workaround would be to work git tagging into a git commit hook looking for changes on a version file.

Thanks for the quick response.

closed 2018-01-19 by sielicki