csete/gpredict #121: {Windows] _send_rigctld_command SIZE ERROR 13/12

wgaylord, 2018-02-05:
This error happens repeatedly till the device is disengaged automatically. 
This seems to be an error with https://github.com/csete/gpredict/blob/420a2be60207f55a097b944fca6ac4b54a1d0e0c/src/gtk-rig-ctrl.c#L1381

csete, 2018-02-05:
Please provide a description of your setup, in particular what kind of server you are using.

wgaylord, 2018-02-05:
Currently it’s just a server that receives all that is sent then sends back a newline as the required return. 

This error is caused by the code in gPredict thinking it sent 12 bytes when it actually sent 13.

wgaylord, 2018-02-05:
NVM apparently not returning "RPRT 0" some how made the code that had nothing to do with the return value error. :/  Very weird that because I did not send back "RPRT 0" it decided it send 13 byte when it should have been 12.

wgaylord, 2018-02-05:
Sorry.  About the spam misclicked the hit reopen.

closed 2018-02-05 by wgaylord