csete/gpredict #125: AC_PROG_INTLTOOL (0.21)

Willamid, 2018-03-19:
Hey,
I'm trying to install gpredict using Cygwin on Windows 7 and I'm getting the following error:

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... no
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make supports nested variables... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... none
checking dependency style of gcc... none
checking for library containing strerror... none required
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
./configure: line 4240: syntax error near unexpected token `0.21'
./configure: line 4240: `AC_PROG_INTLTOOL(0.21)'

```
I tried reinstalling my intltool package but that didn't fix the problem. Any thoughts on how I can fix this?

Thanks in advance

csete, 2018-03-25:
I have no idea, sorry. Do you think it is a bug in the build script?

Willamid, 2018-03-25:
Well if there is actually no syntax error for other users then there must be something wrong with my Cygwin libraries and dev packages. I'll try reinstalling the whole thing and let you know.

csete, 2018-07-02:
I don't know whether there is anything for us to do here?

closed 2018-10-16 by csete