csete/gpredict #126: Now drop-down menu in windows

markusschlotterer, 2018-03-26:
Hi,

just downloaded 2.2.1. The menu bar is there, but the drop-down menus do not show

Regards
Markus

csete, 2018-03-26:
Hello Markus,

I have no idea how this can happen and you are not giving me much information to work with...

If this was a general issue, we would have found it during release testing or shortly after release, as there are now hundreds of people using gpredict on windows. But this is the first time I hear about this, so we will need more info.

You can try to run the gpredict-con and see if whether any errors are printed in the console.

markusschlotterer, 2018-03-27:
Hi Alexandru,

I found the problem. I was using gpredict over remote desktop. That worked fine for gpredict 1.4, but not for >2.0. Maybe a GTK problem? Thanks for your help and sorry for bothering you.

Regards
Markus

closed 2018-03-27 by markusschlotterer

csete, 2018-03-27:
Thanks for the follow-up, Markus. It's good to have this info for future reference.

We have updated the Gtk+ toolkit and it could be that it requires something more than what was included.