csete/gpredict #127: Transponder frequency calibration

prstoetzer, 2018-06-06:
For full Doppler tuning on amateur radio transponder satellites, a calibration feature is needed to apply an offset to the calculated Doppler shift on the uplink, downlink, or both. This is required due to the frequency inaccuracy on both the satellite transponders and users transceivers. The calibration needs to be able to be performed during the pass and separate offsets should be able to be stored for each satellite.

EA4ZK, 2019-12-16:
That is what I was looking for!!! Great issue for the future!!!  Thanks

mripv6, 2020-04-21:
I agree - this is much needed feature. I just attempted to use the radio control feature with my IC9700. It worked great (hamlib 4), but there isn't a way to account for the frequency inaccuracies identified by the OP. 

For the beacon, the calculated frequency is different than the actual frequency. However, I can tune the main dail to find the beacon and then the doppler correction keeps the beacon in tune. However, when I look for my downlink signal, it usually is out of the audio passband. Then, there isn't any way (that I know of) to correct for the offset and let the computer continue to correct for doppler.

I know the solution needs to be generic. With the IC9700, it's possible to press Main or Sub and adjust the uplink or downlink independently to adjust for the offsets. Once the offset is discovered during a pass, then I need another button in Gpredict to apply doppler correction to the new frequencies. (and store the offset) I thought that's what the L button was supposed to do, but it doesn't.

dl7oap, 2020-04-21:
Hi, 
I would also prefer a function like this.
I currently have 2 workarounds to fix the problem
a) i start on a SSB satellite the tracking and press T to be on the mid of the transponder.  i do not use L. i open a new linux shell and make a manual second access to the ic9700 with "rigctl -m 3081 -r /dev/ic9700a -s 19200" and set the RIT with command "J -100" or whatever the needed difference for cw or ssb.
b) i start on a SSB satellite the tracking and press T to be on the mid of the transponder.  i do not use L. and i only change the frequency on the uplink in the radio control GUI with the small triangles at 100hz or 10hz.  but every time you use T or start a new satellite you have to calibrate again.

on heavy load, when updating all 100ms or 300ms, its sometimes even impossible to change the rit manually on the ic9700 because the ic9700 is busy with the incoming command via hamlib. maybe there is also a function need, that minimize unnecessary command via CAT. so there is no need to send a new frequency when it was only changing 10 hz.. but maybe 50 hz or on FM it could be even more.

i am using gpredict 2.3.58.

csete, 2020-05-16:
Hi guys,
Thanks for your input and feedback. As you may have noticed, I have added the label "feature" to this issue, which means that it is on the list to be implemented. I am currently trying to figure out what to do about the radio controller functionality because there are many challenges using the generic hamlib interface. I am afraid we have to rethink the whole concept and fix it once and for all.

dl7oap, 2020-05-16:
Hi, just **brainstorming** to this story:

the offset is different for the mode you are using on a satellite. so working on XW-2F in SSB i need a offset near 300-500 hz. on cw i need 800-1000Hz. maybe you can keep it simple and there is only one offset per satellite and the user have to manage this by himself. so IMHO in a first step i only would implement one offset correction per satellite and then wait some time to see if the users are happy with this.

A) one solution is the variant a) i mention above and which i am using here at home as a workaround. so you could implemented a changeable offset in the radio gui  which will use hamlib command J/j to set the RIT of the radio and will tune the uplink. this will work, but IMHO this will lead to more hamlib commands and therefore to more heavy traffic on the cat interface. normaly every radio which can handle RIT, should work with J/j command via hamlib. 

B) this is the variant b) i mentioned above. here you can, too, implement a changable offset in the gui. but it is added/subtracted from the uplink frequency of gpredict. so here you will save unnecessary cat command, because the correct will just reach the radio, via a internal correction.
i personally would prefer variant B)

just a mockup of how the gui could be changed, so you can change the offset (which is stored per satellite) via plus and minus. Or you even could implement a radio button (a dial knob). 
maybe even a keyboard shortcould like PageUp, PageDw will help the user, to easly correct the offset, when its needed during a satellite pass.

[image]

IMHO the variant b should work for all radios via hamlib, without deep changes in gpredict?

73, de DL7OAP, Andy

mripv6, 2020-05-17:
Thanks for the brainstorming. I think variant B would work. (Should the offset add/subtract from the downlink or uplink?) I don't think it would be difficult with hamlib because the offset is calculate by gpredict. I find that I can use the gpredict control panel radio to make the corrections. Once I figure out the offset, gpredict seems to correct for doppler as I move around the passband using the knob on my IC9700. Either way, for planning/development purposes, the offset needs to be stored with the satellite info and depend on mode. 

I also wonder if this information could be shared between users. Similar to the doppler.sql file in satpc32. That might give some people a starting point. I know every rig is different, but I find similar offsets as DL7OAP.

EA4ZK, 2020-05-17:
Remember that the calibration depends of the rig too, so the same rig could
have different calibration.

whattwood, 2021-04-17:
I would like to 2nd this feature request! Seems like it would simply take 1 more field in the radio control window to make the "fine tune" adjustment. And obviously it would be great if the setting saved for future use on the same satellite.

mdblack98, 2021-04-17:
You'll find --twiddle-timeout=[secs] will allow you to move the VFO knob and -- with the latest hamlib as of yesterday gpredict will resume wherever you stop.  Previously it was a bit of a crap shoot whether it would resume at your current point or snap back to it's own Doppler.  That's what was fixed yesterday.

Rather than using RIT which is not capable on all rigs just adjust the frequency on the RX VFO which, with twiddle-timeout can be done from the VFO knob or the frequency controls in gpredict.
Mike W9MDB

whattwood, 2021-04-17:
Thanks, that sounds like a great feature on hamlib.

However, I'm not sure it will address the issue in question here because,
although gpredict will "obey" manual VFO adjustments, it will also change
the downlink frequency by a correlating amount. What we need is a way to
temporarily unlock the downlink/uplink pair, then re-lock at a slightly
different spot. That way once we've found our downlink on the satellite we
can tune around to different frequencies in the passband and our uplink
will be at exactly the right place.

The workaround for now is to:
1. Unlock the uplink/downlink with the "L" button and, while transmitting
audio, figure out how much to move the TX frequency up or down - usually
between 100 to 500 Hz.
2. Re-lock the uplink/downlink with the "L" button (which pulls the TX
frequency back to the place defined in the trsp file) and move to another
spot on the transponder where there is a station you want to contact.
3. Press the "L" button to unlock uplink/downlink and move TX frequency by
the same amount you found in step 1.

However, the workaround is fairly clumsy when the DX station is drifting or
if you are quickly chasing a number of stations. As you probably know, a
satellite pass can be quite frantic :-)

Cheers,
Bill