csete/gpredict #128: Rotator controller doesn't work on Windows

csete, 2018-06-15:
https://community.libre.space/t/gpredict-hangs-when-engaging-antenna-control-windows/1950/10

csete, 2018-07-06:
Fix applied. Will close issue when we have a windows binary.

tomvdb, 2018-11-10:
Has a new binary been built with this fix by any chance? Latest windows release binary giving me issues on rotator control and I'm not sure if it's this or something else.

csete, 2018-11-10:
> Has a new binary been built with this fix by any chance?

I know my English is not very good, but exactly which part of "Will close issue when we have a windows binary" do you not understand?

tomvdb, 2018-11-12:
Fair enough, just checking since it's been a while since that comment and was just excited about a new controller i'm making :) I've used the version under linux and it works perfect.

Thanks for the reply

W3AXL, 2018-12-02:
Is there any good way to compile the newer versions on our own for windows? I could really use rotator control for our club and I've been trying all evening to compile using MSYS2/Mingw to no avail.

Alternatively is there any ideas on how soon 2.4 builds for windows will be out?

csete, 2019-01-18:
@tomvdb  @W3AXL please try https://github.com/csete/gpredict/releases/download/v2.3/gpredict-win32-2.3.37.zip

Sorry for the late build.

closed 2019-01-19 by csete