csete/gpredict #141: Top menu options not available

arrooney, 2018-07-19:
I have a new download of Gpredict 2.2.1 on my Ubuntu 16.04 xenial desktop. I need to configure my rotator controls, but when I run Gpredict, there is no top bar with the 'Edit' tab that I (am pretty sure) that I need to configure my rotators. 

Attached is a screenshot of my GUI as it appears with no top bar.

[image]
Thanks for any help.

GoatMon, 2018-07-27:
Looks like you are in full screen mode. Hit the down arrow at the top right -> select "Exit Full Screen", Then down arrow -> attach module.

csete, 2018-07-28:
Hi @arrooney 

Sorry for the late reply, but this is a bug and feature request tracker used for development, not a user support forum, which we run through [other means](http://gpredict.oz9aec.net/User_Support.php).

In this case, I suspect what you observe is a problem with the Ubuntu Desktop and not gpredict itself. Or you might be running in full-screen mode as suggested by @GoatMon. It also seems like you are running some specific build of gpredict as there is no GPS support in any of the official builds.

closed 2018-07-28 by csete