csete/gpredict #144: gtk_menu_popup() is deprecated since GTk+ 3.22

csete, 2018-10-17:
See for example:

```
gtk-sat-list-popup.c: In function ‘gtk_sat_list_popup_exec’:
gtk-sat-list-popup.c:83:5: warning: ‘gtk_menu_popup’ is deprecated: Use '(gtk_menu_popup_at_widget, gtk_menu_popup_at_pointer, gtk_menu_popup_at_rect)' instead [-Wdeprecated-declarations]
     gtk_menu_popup(GTK_MENU(menu), NULL, NULL, NULL, NULL,
     ^~~~~~~~~~~~~~
In file included from /usr/include/gtk-3.0/gtk/gtklabel.h:34:0,
                 from /usr/include/gtk-3.0/gtk/gtkaccellabel.h:35,
                 from /usr/include/gtk-3.0/gtk/gtk.h:33,
                 from gtk-sat-list-popup.c:29:
/usr/include/gtk-3.0/gtk/gtkmenu.h:138:9: note: declared here
 void    gtk_menu_popup    (GtkMenu        *menu,
         ^~~~~~~~~~~~~~
  CC       gtk-sat-module.o
  CC       sat-cfg.o
  CC       sat-pref-list-view.o
```

f1ixx, 2019-02-10:
Hi
Work for me.
I've replace :
`(GTK_MENU(menu), NULL, NULL, NULL, NULL,[...]);`
to
`    gtk_menu_popup_at_pointer(GTK_MENU(menu), NULL);
`
I don't test every where in code but work for this.
Hope help

Boris, F1IXX

note : sorry, don't know how make patch

csete, 2019-02-10:
Thanks for the pointer. Unfortunately, this function is only available since version 3.22, but I would like to keep gpredict compatible with all supported versions of Ubuntu.

wkitty42, 2019-05-29:
are you following the ubuntu support timeline? by that i mean where LTS versions are supported until they are out of LTS and no longer supported by ubuntu? the same could be done for the intermediate releases that are not LTS... maybe even continue support for maybe up to 6 months after they are no longer supported? that's basically what other projects are having to do...

csete, 2019-05-29:
@wkitty42 I don't understand what you mean.

I want to keep gpredict compilable on all versions of Debian, Ubuntu, Fedora, ... , that are still supported either by a company or a community. That includes LTS versions of Ubuntu.

wkitty42, 2019-05-29:
what i meant was that, for example, ubuntu 12.04 LTS is long out of support... you're not supporting it still, are you? so basically i was saying that if you follow the ubuntu support timeline, it might make things easier on that front...

i fully understand the support problem... i'm still on 14.04 here and the other day a project had to do the following because it isn't supported in my 14.04 gcc...

> Ah this is a compiler issue: convertibility of std::unique_ptr<T> to T* automatically. We evidently can’t rely on this conversion, so we should add an explicit .get() to fix this. Will do that shortly.

this isn't the first time that project has had to do something similar because of older gcc on older *nix...

anyway, it was just my two centavos from the peanut gallery... i hope it helps a little bit...

csete, 2019-05-29:
> what i meant was that, for example, ubuntu 12.04 LTS is long out of support... you're not supporting it still, are you?

No, I am neither supporting 12.04 nor 14.04.

The suggested solution requires Gtk+ 3.22 first introduced in Ubuntu 18.04 and would thus prevent gpredict to run on 16.04, which is still supported.

wkitty42, 2019-05-29:
ok, so you are following the support timelines as i asked about... good deal! :smile_cat:

commit aac0beae (2020-04-04): Avoid using deprectated gtk_menu_popup in Gtk 3.22+

Thanks to @f1ixx for the tip. Fixes #144.
closed 2020-04-04 by csete