csete/gpredict #146: gdk_screen_width() and gdk_screen_height() are deprecated

csete, 2018-10-17:
See for example:

```
gtk-sat-module-popup.c: In function ‘module_window_config_cb’:
gtk-sat-module-popup.c:1215:9: warning: ‘gdk_screen_width’ is deprecated [-Wdeprecated-declarations]
         x > gdk_screen_width() || y > gdk_screen_height())
         ^
In file included from /usr/include/gtk-3.0/gdk/gdk.h:50:0,
                 from /usr/include/gtk-3.0/gtk/gtk.h:30,
                 from gtk-sat-module-popup.c:26:
/usr/include/gtk-3.0/gdk/gdkmain.h:108:6: note: declared here
 gint gdk_screen_width  (void) G_GNUC_CONST;
      ^~~~~~~~~~~~~~~~
gtk-sat-module-popup.c:1215:9: warning: ‘gdk_screen_height’ is deprecated [-Wdeprecated-declarations]
         x > gdk_screen_width() || y > gdk_screen_height())
         ^
In file included from /usr/include/gtk-3.0/gdk/gdk.h:50:0,
                 from /usr/include/gtk-3.0/gtk/gtk.h:30,
                 from gtk-sat-module-popup.c:26:
/usr/include/gtk-3.0/gdk/gdkmain.h:110:6: note: declared here
 gint gdk_screen_height (void) G_GNUC_CONST;
      ^~~~~~~~~~~~~~~~~
  CC       gtk-sat-module-tmg.o
  CC       gtk-sat-popup-common.o
  CC       gtk-sat-selector.o
  CC       gtk-single-sat.o
```

f1ixx, 2019-02-10:
Hi,
to remove warning (don't know for multi monitor) :
before the test (at line 1218 for me) add :
```
GdkRectangle workarea = {0};
gdk_monitor_get_workarea(gdk_display_get_primary_monitor(gdk_display_get_default()), &workarea);
```

and change the variable in test line :
```
gdk_screen_width() => workarea.width
gdk_screen_height() => workarea.height
```

Hoping to have helped a little
Best 73's

Boris, F1IXX

csete, 2019-02-10:
Hi Boris,

Thanks for the tip! I will take a look and also test it with multi-monitor setup.

Alex, OZ9AEC

commit 994af746 (2020-04-04): Replace deprecated gdk_screen_width() and height()

Thanks @f1ixx for the tip. Fixes #146.
closed 2020-04-04 by csete