csete/gpredict #151: implicit declaration of function timegm

csete, 2018-11-18:
Since the fix from PR #150 I have a compile time warning with 5.4 (xubuntu 16.04):

```
sat-info.c: In function ‘epoch_to_str’:
sat-info.c:240:13: warning: implicit declaration of function ‘timegm’ [-Wimplicit-function-declaration]
         t = timegm(&tms);
```

Including `time.h` makes no difference.

Th code still builds and runs fine.

kukabu, 2018-11-20:
Hello

from the manual
```
   Feature Test Macro Requirements for glibc (see feature_test_macros(7)):

       timelocal(), timegm():
           Since glibc 2.19:
               _DEFAULT_SOURCE
           Glibc 2.19 and earlier:
               _BSD_SOURCE || _SVID_SOURCE
```

csete, 2018-11-20:
Thanks, @kukabu. I will have to read the man page to understand what this means.

commit de0c01cd (2019-01-01): Ensure timegm() is available

Fixes #151.
closed 2019-01-01 by csete