csete/gpredict #153: Rotator Movement Tolerance not observed when in overwind

darksidelemm, 2018-11-30:
My rotator controller is a Rot2Prog, which when used via HamLib, will report azimuth positions outside the 0-360 degree range when in an overwind region. My rotator is setup to be able to spin 720 degrees (i.e. '-180' degrees through to '+540' degrees), and these are the values that hamlib will report when the current position is requested.

When in one of these regions, the position tolerance comparison here will fail, resulting in the rotator being commanded to move every cycle period: https://github.com/csete/gpredict/blob/master/src/gtk-rot-ctrl.c#L846

I've solved a similar issue in SatNOGS by taking the float modulo 360.0 of the current reported azimuth. Since this doesn't seem to be a thing in C++, one option is the following:
```
        /* Handle a rotator reporting an azimuth outside of 0-360 degrees */
        if (rotaz < 0.0){
            rotaz += 360.0;
        } else if (rotaz > 360.0){
            rotaz -= 360.0;
        }
```
This makes the assumption that the reported position will never be outside of -360 to +720, which is probably safe. I am currently running gpredict with this included, and it seems to work OK.

csete, 2018-11-30:
Thanks, @darksidelemm, we'll fix it for the next release.

csete, 2018-12-22:
@darksidelemm,
Can you check whether the code in the current master is OK? commit 65568a9 implements a slightly modified version of your fix.

commit 65568a98 (2018-12-22): Ensure rotator Azimuth is always between 0-360

Fixes #153.
closed 2018-12-22 by csete