csete/gpredict #154: Segmentation fault on Xubuntu 18.04

Yume303, 2018-12-06:
The latest version (from repo/from git too) opens for a second then closes with 'segmentation fault' :/
Shall i provide some gdb info?

csete, 2018-12-06:
Ehm, yes please, provide all the info you can, including whether data has been updated or modified. I run gpredict on Xubuntu 16.04 and 18.04 every day.

Yume303, 2018-12-06:
Last worked at cca Dec. 04., first is suspect some update after that.
Hw: Dell m6700 3rdgen intel, with nvidia, tried to rollback to nouveau and earlier versions, doesn't help on that situ. Short debug:
gdb) run
Starting program: /usr/bin/gpredict 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8bb3700 (LWP 20177)]
[New Thread 0x7fffe3fff700 (LWP 20178)]
[New Thread 0x7fffe37fe700 (LWP 20179)]
[New Thread 0x7fffe2dc0700 (LWP 20180)]

Thread 1 "gpredict" received signal SIGSEGV, Segmentation fault.

Anything else needed?

csete, 2018-12-06:
When you get the segfault in gdb, can you type "bt" to get a backtrace?
Thanks.

csete, 2018-12-06:
Btw. I don't think it is graphics related. Probably some TLE data makes it sad.

Yume303, 2018-12-06:
Thread 1 "gpredict" received signal SIGSEGV, Segmentation fault.
0x00005555555961f5 in ?? ()
(gdb) bt
#0  0x00005555555961f5 in  ()
#1  0x000055555558f856 in  ()
#2  0x00007ffff5ddbc73 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff5ddb1f5 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff5ddb5c0 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff5ddb8d2 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff6e9c385 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#7  0x000055555556542f in  ()
#8  0x00007ffff5483b97 in __libc_start_main (main=0x555555565120, argc=1, argv=0x7fffffffdf08, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdef8) at ../csu/libc-start.c:310
#9  0x000055555556574a in  ()

Latest time it worked, I tried to track ESEO with the preliminary tle...

csete, 2018-12-06:
Unfortunately, there is not much useful information in the backtrace :(

You could try to remove some or all the data from ~/.config/Gpredict/satdata/

Yume303, 2018-12-06:
Still the same :(
Also removing the full config directory works... strange... but it's okay for now :)

Yume303, 2018-12-09:
We can close, if it happens again, I will save and give my full config to investigate :)

closed 2018-12-09 by csete