csete/gpredict #156: Gpredict hangs when attempting to use a TLE with a zero eccentricity.

darksidelemm, 2018-12-16:
The following TLE (created during pre-launch of the SSO-A mission) causes gpredict to hang badly:
```
SSO-A
1 70003U 18835A   18336.80285382  .00000000  00000-0  00000-0 0    05
2 70003  97.8442  45.4226 0000000 100.4497 180.6751 14.96286186    03
```

In fact, I have to kill -9 to be able to exit out of gpredict. This bug is reproducible, just try and use the above TLE. Probably something to do with the zero eccentricity term.

csete, 2018-12-16:
Confirmed and definitely a bug. It is indeed the 0 eccentricity that makes it malfunction.

A temporary workaround suggested for those who can build from source:

```
diff --git a/src/sgpsdp/sgp_in.c b/src/sgpsdp/sgp_in.c
index 5a23d54..c7e3e16 100644
--- a/src/sgpsdp/sgp_in.c
+++ b/src/sgpsdp/sgp_in.c
@@ -214,6 +214,8 @@ Convert_Satellite_Data (char *tle_set, tle_t *tle)
        strncpy (&buff[1], &tle_set[95], 7);
        buff[8] = '\0';
        tle->eo = g_ascii_strtod (buff, NULL);
+    if (tle->eo == 0.0)
+        tle->eo = 0.0001;
 
        /* Satellite's Argument of Perigee (degrees) */
        strncpy (buff, &tle_set[103], 8);

```

closed 2018-12-22 by csete

kerel-fs, 2021-09-01:
I just re-discovered this bug while trying to load a pre-launch TLE for Qubik.
While fixed in master, there was no release of gpredict done since this was fixed in Dec 2018.

commit 79e31b27 (2018-12-22): Prevent divisin by 0 when eccentricity is 0

Fixes #156.