csete/gpredict #158: OSX support

Fohlen, 2018-12-27:
Howdy dear `gpredict` team

I was able to compile `gpredict` successfully under OSX using the following adjustments:

- I had to change `libtoolize` to `glibtoolize`as described [here](https://stackoverflow.com/a/15448876/3135319)
- I had to force link `gettext` since it contains the `nls` macro (see [here](https://stackoverflow.com/a/44960701/3135319)) using `brew link gettext --force`
- I had to install `goocanvas` via `brew install goocanvas`

After this I executed:

```
./autogen
./configure --prefix=/Users/fohlen/gpredict
make -j4
make install
```

successfully. Now I do have the following folder structure:

```
/Users/fohlen/gpredict
/bin/
  ./gpredict
/share/
  applications/ gpredict/   locale/   man/   pixmaps/
```

When starting the application I get the following error message:

```
(process:30634): Gtk-WARNING **: 13:53:53.460: Locale not supported by C library.
	Using the fallback 'C' locale.
```

The window opens up for the fraction of a second after it closes again (without any further notice).
What is going wrong here and how could I fix it?

Fohlen, 2018-12-27:
I can report back: it actually works. I have done a local install to `/usr/local` and now it works. However to build this thing officially OSX instructions are a bit "missing" and also the `autogen.sh` should be adjusted but that's not to hard.

Fohlen, 2018-12-27:
Also I noticed that apparently there is already a working `brew install gpredict` package, however adding official support for it here would be cool 👍

csete, 2018-12-27:
Okay, I'm a little confused... Is this a bug report or? If yes, how does MacPorts and homebrew handle it?

Not sure what you mean with official support compared to the readily available packages, but unless you or somebody else is volunteering to set it up **and** maintian it, I am not interested in discussing it.

Fohlen, 2019-01-02:
@csete I am unsure where the patch to the `autogen.sh` is stored for brew. With official I mean: it would be good to have the build instructions and dependencies for OSX known so someone can try to build themselves. I'd be willing to submit these in a PR.

csete, 2019-01-05:
You are welcome to submit a PR; however, please check [homebrew formula](https://formulae.brew.sh/formula/gpredict) first because it looks as they can build without any modification of the source package.

closed 2019-01-05 by csete