csete/gpredict #160: Satellite priorities

csete, 2019-01-18:
The autotrack feature could be enhanced if the user could assign priorities to the satellites in a module. See e.g. https://community.libre.space/t/antenna-control-always-track-closest-satellite/1749/5

Mark-H-W, 2023-02-14:
I work for the Space Science and Engineering Lab at Montana State University. This is a feature that we need for tracking multiple satellites, so I went ahead and implemented it in this fork: [https://github.com/Mark-H-W/gpredict](https://github.com/Mark-H-W/gpredict). I know new features are not high on the priority list right now so I have not sent a pull request yet. Let me know if I should send one, or if there are any changes I should make first.

noaa-rfss, 2023-09-13:
@Mark-H-W This is fantastic and exactly what I was for for tracking ARGOS sats.  You even helped with the next big question which is priority - thanks for that and nice work! 

2 follow up questions for you:

1. Was hoping that if gpredict closed or needed to be restarted that the autotrack functionality and priority settings would be restored on recovery.  Is this not the case?  I'm probably missing something simple.
2. For the priorities, I see 0 through 10.  I assume 0 is the highest priority and 10 the least, is this correct?  Also, can I have multiple sat the same low priority i.e. all 7 since I dont care about them unless I'm not tracking anything else.

Thanks again for the help!
Jason

Mark-H-W, 2023-12-21:
@noaa-rfss, apologies for the delayed response.

1. Currently, I have not implemented code to save the autotrack priorities. This means when gpredict is closed the priorities will be reset to zero.

2. A priority setting of 0 is the least priority and 10 is the highest. You can have multiple satellites set to the same priority. It should track the satellite that appeared first if there are multiple satellites with the same priority visible.

I hope this helps,
Mark