csete/gpredict #164: "Show next pass" doesn't always show the next pass

Vinz87, 2019-05-05:
If I right-click on a satellite in whatever view and I select "show next pass", sometimes the pass selected in the pop-up view is not the next one but the subsequent one.
[image]
[image]

Other times it shows the actual next pass, the same listed in the previous view, as I think is the intended behaviour.

csete, 2019-05-06:
Hi,

Please note that when predicting passes, only passes that will have a minimum elevation above a certain threshold will be listed, see Preferences -> Predict -> Minimum elevation. This setting does not apply for "Next" in the real-time views. So I think this is just a duplicate of issue #75.

Also, in the future when reporting bugs, please make sure to include a thorough description of the environment, including which version of gpredict you are using and how it was installed.

Vinz87, 2019-05-06:
You're right, it was the "minimum elevation" which prevented the pass from being displayed.

csete, 2019-05-06:
Okay then, we will be fixing this in issue #75, so I am closing this one.

closed 2019-05-06 by csete