csete/gpredict #165: Incorrect decayed status in list view

csete, 2019-05-30:
See https://github.com/csete/gpredict/blob/master/src/gtk-sat-list.c#L612 however, changing the value to decayed(sat), 0, or FALSE prevents anything to be rendered.

wkitty42, 2019-05-30:
i looked at that and the decayed() routine but i'm not a C coder so i don't understand it very well...

your comment about changing the value almost seems to indicate that once it is set for one satellite, it is not reset to the default for the next satellite...

FWIW: i am a 35+ year coder... i can read some C/C++ and make minor adjustments but my base is ASM and Pascal plus numerous of today's languages (eg: perl, php, sql, java, javascript, etc)...

csete, 2019-05-30:
The decayed() function is fine. It's the graphical widget or the way we use it that is horribly broken. I spent 4 hours trying the most obvious fixes but they didn't help.

wkitty42, 2019-05-31:
On 5/30/19 2:38 PM, Alexandru Csete wrote:
> The decayed() function is fine. It's the graphical widget or the way we use it 
> that is horribly broken. I spent 4 hours trying the most obvious fixes but they 
> didn't help.

ahhh, ok... i was just going by the description of the attempt to fix and the 
resulting failure to render the sats... i've danced that dance before and never 
liked it...

i'll be quiet and simply post more bugs on the forum as i find them ;)