csete/gpredict #169: Closing Antenna Control Closes Application

ghostop14, 2019-06-05:
When the antenna control window is open, closing the antenna control window terminates gpredict.

csete, 2019-06-07:
Details please.

ghostop14, 2019-06-07:
Hi Alexandru, sure, no problem.  Thanks for looking into it.  Details:

Latest gpredict from git, branch: master 32050ce.  
Running on Ubuntu 16.04 LTS 64-bit fully up to date.  
An antenna is configured to point to localhost as a net.

Steps to reproduce: Run gpredict.  Select Antenna Control.  Don't have to connect the rotator or click track.  Just opening the window is sufficient.  Close the Antenna Control Window.  The whole gpredict application exits.

lrowen, 2019-12-09:
I am also intermittently experiencing this issue.  I have not found a pattern to reliably reproduce, but it does seem to only happen when rotator control has actively been engaged.
Running latest git version of gpredict on CentOS 7 x86_64 fully updated.

csete, 2026-08-16:
I have recently fixed issues similar to this, so I hope it is OK now.

closed 2026-08-16 by csete