csete/gpredict #170: Table view doesn’t save all selected settings

wkitty42, 2019-06-06:
when looking at the table in a module, i always click on the “Next Event” header to sort the list by order of the next AOS… then configuring the module and saving the changes, this selection is lost and i have to click on it again…

there’s the same problem with the autotrack setting for the module, too… it has to be reselected after the refresh…

csete, 2019-06-07:
Both the column and sort order should be saved. I can't investigate further without knowing the details of which version and what kind of environment the software is running on.

wkitty42, 2019-06-08:
On 6/7/19 1:57 PM, Alexandru Csete wrote:
> Both the column and sort order should be saved. I can't investigate further 
> without knowing the details of which version and what kind of environment the 
> software is running on.

right now the system is running:

gpredict 2.2.1-ubuntu14.04~ppa2
kubuntu 14.04.6 LTS

i'm currently using the four pane layout but i'm not sure that matters... the 
table of satellites at the bottom doesn't always the currently selected column 
sort... the autotrack option in the drop menu on the top right pane also doesn't 
get saved and restored... every time i adjust the configuration and the panes 
reload, the sort of the columns is reset and the autotrack is unselected...

someone had told me that if i select the sort column and close the pane, that 
the option will be saved... i had closed gpredict the other day and it has come 
back up with the sort order that i had selected and the autotrack was unselected...

while writing this message i did the following:

1. selected the catnum column
2. selected autotrack
3. went into configure
4. clicked the [OK] button to exit without changing anything

the column was reset to the Next Event which i had selected the other day and 
autotrack was unselected again...

csete, 2019-06-08:
Please don't mind my first comment, which I have already deleted before you replied. I was just too stressed and forgot that I have already looked into this a little as mentioned in the [initial thread](https://community.libre.space/t/table-view-doesnt-save-all-selected-settings/4021/2). Sorry for the confusion.

wkitty42, 2019-06-08:
it's alright... we're cool :)