csete/gpredict #171: Low Volume on Icom PCR-100 receiver.

magellan-13016, 2019-06-07:
Hi,
I'm using GPredict 2.2.1 on linux and 2.3.37 on Windows to control my Icom PCR-100 receiver and make Doppler correction with HAMLIB.

I'm using this command line to make the link between the PCR-100 and GPredict :+1: 
rigctld -v -m 402 -r /dev/ttyUSB0 -s 38400 -t 4532 -C stop_bits=1

All is working OK but when GPredict turn on the receiver, the output volume is very low compared to PCRAnywhere software.

I think it would be nice to be able to select the output volume/gain of the receiver because when the link is made between GPredict and the hardware, there is no way to control it since we can only make one link at a time and it is used by GPredict.

I'm teacher and using GPredict for ISS educational project (have a look at the press review at www.f4bpp.com).

Thanks for your help.

David.

csete, 2022-07-16:
Sorry you are having this issue, but it's outside the scope of gpredict to control such radio specific issues.

closed 2022-07-16 by csete

mdblack98, 2022-07-17:
What if you use PCRAnywhere -- then leave the radio on and start gpredict.  Is the audio OK then?
Does the rig have a default audio setting on power up maybe?
You should be able to do this after rigctld is running to turn the volume all the way up.
rigctl -m 2 L AF 1.0
The 1.0 is a range from 0-1.
Mike W9MDB

magellan-13016, 2022-07-19:
Thank you Alexandru and Michael for your response but I regret to say that you are both wrong. In fact, when using the Hamlib library, there are parameters to pass to adjust the volume of the transceivers and if the parameters are not good or missing, you end up with unusable audio (sound too weak, bad mode, squelch too high...).
So I made myself a little program that allows to start the icom PCR-100 via Hamlib so that Gpredict can take control of it while allowing me to adjust the audio level and the squelch in order to be able to decode the SSTV images from the ISS taking into account the Doppler effect. I added two buttons in my program for this purpose: One to launch Gpredict and the other to launch QSSTV which is my favourite software for decoding the images.

[image]