csete/gpredict #172: Gpredict crashes on some predictions

reald, 2019-07-21:
See attached image. 

[image]

In the list for NOAA 15 the next AOS is predicted for 11:58:57. (1) Click on "Nächsten Überflug anzeigen" ("show next pass", 2) => gpredict crashes. Interestingly heavens-above (see background) does not predict this pass.

Next problem: Open future passes table (3). Right click on the prediction for 16:47:22. Only "Details anzeigen" (show details) is available. Click on it, gpredict crashes. Heavens-above does not predict this pass, too.

Reproducabilty: Always
Affected Versions: 2.3.56-3205 (recent on github) and 2.1 (Ubuntu 18.04.2)

csete, 2019-07-21:
If you post your ground station coordinates and the TLE data we might b able to reproduce in the future, though I do not have the time to look at it right now before the prediction "expires".
Note that heaven above seems to be set to 10-degree threshold, which might be why it is not predicting the pass.

reald, 2019-07-21:
Ground Station: 49.52°N 11.00°E 0m

```
[Satellite]
VERSION=1.1
NAME=NOAA 15
NICKNAME=NOAA 15
TLE1=1 25338U 98030A   19198.88778160  .00000058  00000-0  43023-4 0  9993
TLE2=2 25338  98.7452 220.1004 0010986 152.6507 207.5253 14.25921617101157
STATUS=0

[Satellite]
VERSION=1.1
NAME=NOAA 18
NICKNAME=NOAA 18
TLE1=1 28654U 05018A   19198.89265270  .00000059  00000-0  57054-4 0  9995
TLE2=2 28654  99.0936 246.2971 0015374  95.9258 264.3666 14.12438361729545
STATUS=0

[Satellite]
VERSION=1.1
NAME=NOAA 19
NICKNAME=NOAA 19
TLE1=1 33591U 09005A   19198.94705508  .00000041  00000-0  47560-4 0  9992
TLE2=2 33591  99.1805 190.7796 0013035 298.8278  61.1584 14.12365435537778
STATUS=0
```

If I set minimum elevation in gpredict to 10 degrees then the 16:47:22 disappears from the future passes list (3) because the max elevation is 5.48°. But it stays in the list under the map (1). If I select next pass (2) there it does not crash anymore but shows the next pass starting at 18:22:52.

csete, 2019-07-21:
Thanks. I'm trying to reproduce it.

reald, 2019-07-21:
Thanks. 

Found an interesting effect: I´m only interested in a rough pass so I modified the pass time resolution to 120s to get fewer table entries. If I set it to 20s it works. Maybe the pass is now too short for some processing.

[image]