csete/gpredict #173: Print report vs real life

SQ5KTM, 2019-07-21:
Alex, I'm user of Your great Gpredict software. Thank You for developing this application

I realised that there maybe a little bug there or I am doing something wrong.
When I choose upcoming passes and make Summary report there are informations about max elevation. Text info shwoing much lower elevation than will be real and also polar view showing right elevation. Just text and Summary report are showing lower elevation parameters. The difference increases with elevation so it's easy noticable at near zenith passes, sometimes about 15 degrees diff.

PS. Gpredict 2.3.37 Windows edition.

csete, 2019-07-21:
I will take a look but it will be a long while before I can work on gpredict again.

SQ5KTM, 2019-07-21:
Ofcourse, take Your time :)

W dniu 21.07.2019 o 13:14, Alexandru Csete pisze: